Dasharo/open-source-firmware-validation #609: Extend audio tets

macpijan, 2024-11-22:
Detect `sound card` and `audio sink` separately.

Currently our detection methods might be different for Linux and Windows. If there is no actual oputput connected, we may have PASS for Linux (we check for sound card presence only), while at the same time we get FAIL for Windows (maybe we check just for sink, or maybe Windows is different in this matter?)

SebastianCzapla, 2024-11-25:
Related: https://github.com/Dasharo/open-source-firmware-validation/issues/593

Commit: 15560f19cde19b4a2d841a26fd196018ff3272ac changes detection on windows side to detect wider spectrum of devices, sources, and sinks, which could be used as a base for separated detection (filtering speakers from audio cards or virtual sound devices)

SebastianCzapla, 2025-01-09:
@macpijan Currently, out autodetection script fills bunch of variables, all of the are in form of AUDIO_DEVICE_1 / 2 / 3

There is also few tests that are currently in the scope, like playback tests, that are not automated. Are we okay with semi-automation of certain tests?

My idea was to introduce set of named variables, like AUDIO_DEVICE_CARD, AUDIO_DEVICE_SINK, AUDIO_DEVICE_MIC, AUDIO_DEVICE_EXTERNAL_MIC. Then each test would detect appropriate device: Audio Subsystem Detection would attempt to detect card, External Headset Audio Capture would attempt to detect AUDIO_DEVICE_EXTERNAL_MIC and so on. Is this a valid solution here? Detection of for example capture device would still not be 100% confirmation of PASS, but failure of such semi-auto test would still be an indicator of errors that need to be fixed.

SebastianCzapla, 2025-03-27:
[audio-subsystem.robot-linux.zip](https://github.com/user-attachments/files/19485800/audio-subsystem.robot-linux.zip)