Dasharo/open-source-firmware-validation #613: e2e tests fail to enroll DPP credentials

PLangowski, 2024-11-27:
### Device

QEMU

### RTE version

N/A

### OSFV version

dts-tests

### Brief summary

The e2e tests that require DPP keys fail to load them properly.

### How to reproduce

Run a test from `dts/dts-e2e.robot` which requires DPP credentials

### Expected behavior

PASS

### Actual behavior

FAIL:
```
No match found for 'Do you want to participate in this project?' in 5 minutes. Output:

 Dasharo Tools Suite Script 2.1.0
 (c) Dasharo <contact@dasharo.com>
 Report issues at: https://github.com/Dasharo/dasharo-issues
*********************************************************
**                HARDWARE INFORMATION
*********************************************************
**    System Inf.: Micro-Star International Co., Ltd. MS-7D25
** Baseboard Inf.: Micro-Star International Co., Ltd. PRO Z690-A WIFI DDR4(MS-7D25)
**       CPU Inf.:
*********************************************************
**                FIRMWARE INFORMATION
*********************************************************
** BIOS Inf.:  1.0.0
*********************************************************
**                DPP credentials
*********************************************************
**       Logs key:
**   Download key: <LOGS_KEY>
**       Password: <DOWNLOAD_KEY>
*********************************************************
**     1) Dasharo HCL report
**     2) Install Dasharo Firmware
**     3) Restore firmware from Dasharo HCL report
**     4) Edit your DPP keys
*********************************************************
R to reboot  P to poweroff  S to enter shell
K to launch SSH server  L to enable sending DTS logs
V to enable verbose mode
Enter an option:
```

The keys are misplaced.

### Link to screenshots or logs

N/A (would leak DPP credentials)

PLangowski, 2024-11-27:
I have found the most likely reason for the issue:
The test chooses the `Load DPP keys` option from the menu and presses Enter. However, after the newly introduced UI/UX changes the DTS menu only requires the option, without Enter. I will test a solution to this and provide the result here.

artur-rs, 2024-11-27:
@PLangowski commit/PR for the changes?

PLangowski, 2024-11-28:
4b97699eed4e2839a1ba3296267b2afc1e41ac37

closed 2024-12-03 by PLangowski