Dasharo/open-source-firmware-validation #625: VP3210 refuses to take dcu-altered rom

wiktormowinski, 2024-12-03:
### Device

Protectli VP3210

### RTE version

-

### OSFV version

branch: vp-3210-testing, (based on develop 4147db463598a5d7a8757667b8f14f1ecac09801) current commit b125dabeefd936b8296e6718c393fd9ccd67b953

### Brief summary

DCU suite fails due to the changes made not being actually implemented

### How reproducible

happens every time to me

### How to reproduce

try running `dcu.robot` suite

### Expected behavior

the test should modify the rom

### Actual behavior

the test fails because the changes made, are not properly implented

### Link to screenshots or logs

[dcu.robot_log-fail.zip](https://github.com/user-attachments/files/17996239/dcu.robot_log-fail.zip)

miczyg1, 2024-12-05:
According to logs:

```
11:46:35.937	INFO	Flash write failed with code 1	
11:46:35.938	INFO	Executing command: flashrom -p linux_spi:dev=/dev/spidev1.0,spispeed=16000 -c MX25L12833F/MX25L12835F/MX25L12845E/MX25L12865E/MX25L12873F -w /data/write.rom
flashrom v1.3.0 on Linux 5.4.69 (armv7l)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

So obviously, if the flashing failed, the changes would not apply.

wiktormowinski, 2024-12-06:
Today it started working just fine, i guess something changed and updating submodules & launching requirements.txt solved this problem.
[dcu.robot_log_pass.zip](https://github.com/user-attachments/files/18038665/dcu.robot_log_pass.zip)

closed 2024-12-06 by wiktormowinski