Dasharo/open-source-firmware-validation #628: Tests fail on platforms that are slow to get an internet connection

filipleple, 2024-12-06:
### Device

OptiPlex 7010, for example

### RTE version

any

### OSFV version

any

### Affected component(s) or functionality

Any tests that try to perform some network action right after booting into OS

### Brief summary

Tests fail on platforms that are slow to get an internet connection, if they try to perform some network action right after booting into OS they try to 

### How reproducible

99%

### How to reproduce

Run a test that tries to perform some network action right after booting into OS, e.g., [tpm-support]().

### Expected behavior

It should test the functionality specified by the test case and fail only if the criteria of said functionality are not met

### Actual behavior

It fails, because the platform is rather slow and it hadn't established a network connection yet

### Link to screenshots or logs

```
==============================================================================
Tpm-Support
==============================================================================
TPM001.001 TPM Support (firmware) :: This test aims to verify that... ......
Chip detection failed, attempting cbmem log detection

TPM001.001 TPM Support (firmware) :: This test aims to verify that... | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 5 times. The last error was: 'ping: g
oogle-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
------------------------------------------------------------------------------
TPM001.002 TPM Support (Ubuntu) :: Check whether the TPM is initia... | FAIL |
AttributeError: module 'requests.exceptions' has no attribute 'ProtocolError'
------------------------------------------------------------------------------
TPM001.003 TPM Support (Windows) :: Check whether the TPM is initi... | SKIP |
TPM001.003 not supported
------------------------------------------------------------------------------
TPM002.001 Verify TPM version (firmware) :: This test aims to veri... ......
Chip detection failed, attempting cbmem log detection

TPM002.001 Verify TPM version (firmware) :: This test aims to veri... | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 5 times. The last error was: 'ping: g
oogle-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
------------------------------------------------------------------------------
TPM002.002 Verify TPM version (Ubuntu) :: This test aims to verify... | PASS |
------------------------------------------------------------------------------
TPM002.003 Verify TPM version (Windows) :: This test aims to verif... | SKIP |
TPM002.003 not supported
------------------------------------------------------------------------------
TPM003.001 Check TPM Physical Presence Interface (firmware) :: Thi... | FAIL |
Keyword 'Check Internet Connection On Linux' failed after retrying 5 times. The last error was: 'ping: g
oogle-public-dns-a.google.com: Temporary failure in name resolution' does not contain ', 0% packet loss'
------------------------------------------------------------------------------
TPM003.002 Check TPM Physical Presence Interface (Ubuntu) :: This ... | PASS |
------------------------------------------------------------------------------
TPM003.003 Check TPM Physical Presence Interface (Windows) :: This... | SKIP |
TPM003.003 not supported
------------------------------------------------------------------------------
Since the asset has been checkout manually by you prior running this script, it will NOT be checked in a
utomatically. Please return the device when work is finished.
Tpm-Support                                                           | FAIL |
9 tests, 2 passed, 4 failed, 3 skipped
==============================================================================
```

### Solutions you've tried

I haven't tried fixing it, but it probably could be resolved with adding some `Wait For Internet Connection` keyword after booting into OS, checking `ip a | grep 192` with a 5m+ timeout

SebastianCzapla, 2025-01-07:
Addressed it in this commit:
https://github.com/Dasharo/open-source-firmware-validation/pull/641/commits/29a4c530e80eb8ab83dd44ff7f896a51eee26b06

It swaps order of keywords in a way that makes more sense; we wait for the moment out DUT is assigned an IP, then we check connection quality with ping.

filipleple, 2025-01-16:
Should be resolved by this commit: https://github.com/Dasharo/open-source-firmware-validation/pull/667/commits/bf653915c31f6c0570b8240f87d11c99fc684d7a

SebastianCzapla, 2025-02-06:
This change https://github.com/Dasharo/open-source-firmware-validation/pull/678 adds 50s timeout for each retry out of 10, and should in theory cover this issue.

SebastianCzapla, 2025-04-17:
Closing as https://github.com/Dasharo/open-source-firmware-validation/pull/678 was merged and should cover the failing scenario