Dasharo/open-source-firmware-validation #646: Power On keywords fail via SSH

SebastianCzapla, 2024-12-20:
### Device

All Novacustom

### RTE version

----

### OSFV version

develop branch

### Affected component(s) or functionality

Power On and similar keywords

### Brief summary

Power on depends on RTE, it fails during Novacustom testing due to SSH

### How reproducible

100%

### How to reproduce

Prepare OSFV repository for testing, on develop

Run a test suite like  dasharo-compatibility/wifi-bluetooth-support.robot

### Expected behavior

Tests run as expected

### Actual behavior

Power On -> "No keyword with name 'Rte Psu Off' found."
Power Cycle On -> "No keyword with name 'Rte Psu Off' found."
Power On Default -> Attempts to pull RTE data from Snipeit, which fails because laptop testing is done without RTE

### Link to screenshots or logs

[image]
[dasharo-compatibility.zip](https://github.com/user-attachments/files/18210866/dasharo-compatibility.zip)

m-iwanicki, 2025-01-02:
Error also happens on main branch and when trying to test `dasharo-compatibility/dasharo-tools-suite.robot` on `config:msi-pro-z690-a-ddr5`:

```
==============================================================================
Dasharo-Tools-Suite
==============================================================================
DTS001.001 Booting DTS from USB works correctly :: This test aims ... | FAIL |
Parent suite setup failed:
No keyword with name 'Rte Psu Get' found.
```

m-iwanicki, 2025-01-02:
@SebastianCzapla I had to recreate python `venv`, after that I had no errors.

SebastianCzapla, 2025-01-07:
@m-iwanicki recreating new venv on freshly cloned OSFV on dasharo branch still fails for Novacustom devices, with the same errors as in the original screenshot.

m-iwanicki, 2025-01-07:
Can you post command you used to run tests and Python version?

SebastianCzapla, 2025-01-07:
```
 RTE_IP=192.168.4.222 DEVICE_IP=192.168.4.222 CONFIG=novacustom-v540tnd ./scripts/run.sh dasharo-compatibility/wifi-bluetooth-support.robot
```

With python 3.12.7

wiktormowinski, 2025-02-25:
@SebastianCzapla Setting `RTE_IP` to `none` and using `novacustom-common.Power On`
Power control seems to work however new problems emerge

here are a few iterations i gave up after a while 
[iterations_powerctrl_novacustom.zip](https://github.com/user-attachments/files/18964726/iterations_powerctrl_novacustom.zip)

JakubRed, 2025-02-27:
I have the same issue.
logs:

[cpu-performance.robot_.zip](https://github.com/user-attachments/files/19011930/cpu-performance.robot_.zip)

philipanda, 2025-02-27:
@m-iwanicki 
> That's most likely (based on logs) becase of:
[open-source-firmware-validation/keywords.robot](https://github.com/Dasharo/open-source-firmware-validation/blob/cd4afc87edc380fc1628aaf0f04edc3b5d8e9437/keywords.robot#L558)
Line 558 in [cd4afc8](https://github.com/Dasharo/open-source-firmware-validation/commit/cd4afc87edc380fc1628aaf0f04edc3b5d8e9437)
     IF    '${OPTIONS_LIB}'=='dcu'    RETURN 
 and
[open-source-firmware-validation/platform-configs/include/novacustom-common.robot](https://github.com/Dasharo/open-source-firmware-validation/blob/cd4afc87edc380fc1628aaf0f04edc3b5d8e9437/platform-configs/include/novacustom-common.robot#L34)
Line 34 in [cd4afc8](https://github.com/Dasharo/open-source-firmware-validation/commit/cd4afc87edc380fc1628aaf0f04edc3b5d8e9437)
 ${OPTIONS_LIB}=                                     dcu 
Prepare Test Suite doesn't import osfv_cli robot keywords in this case.

> osfv_cli also doesn't support novacustom so I guess
[open-source-firmware-validation/platform-configs/include/novacustom-common.robot](https://github.com/Dasharo/open-source-firmware-validation/blob/cd4afc87edc380fc1628aaf0f04edc3b5d8e9437/platform-configs/include/novacustom-common.robot#L138)
Line 138 in [cd4afc8](https://github.com/Dasharo/open-source-firmware-validation/commit/cd4afc87edc380fc1628aaf0f04edc3b5d8e9437)
 Power On 
would need to be fixed.

The laptops should always use `${POWER_CTRL}==none` and `${OPTIONS_LIB}==dcu`. Sadly the two settings were used interchangeably to identify novacustom laptops. That is, for example, that in many places it is assumed that if options_lib is `dcu`, then the `power_ctrl` is `none`, but it doesn't have to be this way. But the real cause of the issue is that the libraries are imported despite not using RTE, because a laptop is being tested. It has to be an invalid configuration, that is one different from  `${POWER_CTRL}==none` and `${OPTIONS_LIB}==dcu`.

@wiktormowinski 
In `/2025_02_24_20_06_49/dasharo-performance/disk-io.robot_log.html` the disk-io test suite is trying to perform a reboot without logging in into the OS. The `Power On` keyword is defined in `novacustom-common.robot` to do nothing. The platform should be on and booted into Ubuntu by default. The keyword `Power Cycle Into Ubuntu` is written in such a way that in this case "Login to Linux" has to be called before it. The test should start with the sequence:
```
Power On
Boot System Or From Connected Disk    ubuntu
Login To Linux
```
The test suite is not on develop right now, so I suppose it's a new one and there's just this simple error in it.

@JakubRed in your case it seems that the platform configs are not imported properly

[image]
The image shows that `${POWER_CTRL}` is not set to `none`, but `sonoff`, but the osfv libraries were not imported properly. Are you testing a laptop, that should have `${POWER_CTRL}` set to `none`? If no, then have you tried running `deactivate`, removing your `venv`, reinstalling all the dependencies, and doing `source venv/bin/activate` again, like @m-iwanicki suggested?

philipanda, 2025-02-27:
@wiktormowinski 
> Setting `RTE_IP` to `none` and using `novacustom-common.Power On`
Power control seems to work however new problems emerge

The `RTE_IP` can be set to anything, because it is not used if `${POWER_CTRL}==none` but it has to be set. It is checked if it is set somewhere along the way, and we should finally remove this check TBH.

As for using the `novacustom-common.Power On`, that is probably the error created in https://github.com/Dasharo/open-source-firmware-validation/pull/637, at https://github.com/Dasharo/open-source-firmware-validation/pull/637/files#diff-2bd69584d51e03632b6b478cae660b9d98acb614bedfcf8ad005f876c9d6744eR1548-R1570.
Check https://github.com/Dasharo/open-source-firmware-validation/issues/703 for details and hotfixes. If you have an idea how to solve this smoothly, then let's discuss under the issue please

mkopec, 2025-02-27:
Fixed by https://github.com/Dasharo/open-source-firmware-validation/pull/704 and https://github.com/Dasharo/osfv-scripts/pull/113  , please test and confirm if it works

SebastianCzapla, 2025-02-27:
@mkopec 
Tested with the command provided by you, as well as with run.sh command, and got successful run.
```
RTE_IP=0.0.0.0 DEVICE_IP=192.168.4.68 SNIPEIT_NO=True CONFIG=novacustom-v540tu ./scripts/run.sh dasharo-compatibility/dmidecode.robot
```

closed 2025-04-09 by BeataZdunczyk