Dasharo/open-source-firmware-validation #650: Too many false negatives in DTS tests

DaniilKl, 2025-01-08:
### Device

QEMU Q35 Emulator

### RTE version

-

### OSFV version

cd4afc87edc380fc1628aaf0f04edc3b5d8e9437

### Affected component(s) or functionality

DTS E2E tests

### Brief summary

Too many false negatives in DTS tests

### How reproducible

Unknown

### How to reproduce

Try to run test suite via `meta-dts` CI.

### Expected behavior

Tests not fail because of some timing issue.

### Actual behavior

Tests fail because of timing issues

### Link to screenshots or logs

For example for MSI:
```
E2E003.005 MSI PRO Z690-A DDR-4 initial deployment (legacy -> Core... | FAIL |
No match found for 'Press Enter to continue' in 30 seconds. Output:
Enter logs key:                x

Enter firmware download key:   x

Enter password:                x
Dasharo DPP credentials have been saved
Checking for Dasharo firmware...
Waiting for network connection ...
Network connection have been established!
Checking if board is Dasharo compatible.
Waiting for network connection ...
Network connection have been established!
```

For PCENGINES:

```
E2E005.002 PC Engines DPP initial deployment (legacy -> Coreboot +... | FAIL |
No match found for 'Press Enter to continue' in 30 seconds. Output:
Enter logs key:                x

Enter firmware download key:   x

Enter password:                x
Dasharo DPP credentials have been saved
Checking for Dasharo firmware...
Waiting for network connection ...
Network connection have been established!
Checking if board is Dasharo compatible.
Waiting for network connection ...
Network connection have been established!
```

Logs acquired from [this CI run.](https://github.com/Dasharo/meta-dts/actions/runs/12652267778)

### Additional context

The places where the tests fail and randomless of the fails suggest that the cause are timings.  But the real cause should be detected before making any decisions or spending time on solving the issue.

### Solutions you've tried

Rerunning CIs. The results are rather random.

m-iwanicki, 2025-01-10:
@DaniilKl Merged fix into develop but issue should probably stay open until couple CI runs to make sure all problems with timeouts were fixed.
Unfortunately my fix might make https://github.com/Dasharo/open-source-firmware-validation/issues/652 problem a little worse especially if all test fail.

artur-rs, 2025-02-06:
@m-iwanicki bump, what is the status with v2.2.1 workflows?

m-iwanicki, 2025-02-07:
This problem appears to be fixed. CI was run many times and issue didn't reappear.

closed 2025-02-07 by m-iwanicki