Dasharo/open-source-firmware-validation #656: Adding remaining workflows and platforms.

DaniilKl, 2025-01-09:


DaniilKl, 2025-01-24:
This list **is in progress**.

## E2E

#### HCL Report

#### Dasharo Deploy

  - [ ] Installation
    - [ ] Community (Coreboot + UEFI) installation 
    - [ ] DPP (Coreboot + UEFI) installation with credentials
    - [ ] DPP (Coreboot + Seabios) Installation with credentials
    - [ ] DPP (Coreboot + Heads) Installtion with credentials
  - [ ] Update
    - [ ] Generic update
    - [ ] UEFI Capsule Update
      - [ ] Community -> DPP (Coreboot + UEFI) without credentials -> Update should not pass
        - [ ] MSI PRO Z790-P DDR4
        - [ ] MSI PRO Z790-P
        - [ ] MSI PRO Z690-A DDR4
        - [ ] MSI PRO Z690-A
        - [ ] Novacustom NV4x 11th Gen Tiger Lake
        - [ ] Novacustom NS5x/7x 11th Gen Tiger Lake 
        - [ ] Novacustom NS5x/7x 12th Gen Alder Lake
        - [ ] Novacustom NV4x 12th Gen Alder Lake
        - [ ] Novacustom V540TU/V560TU 14th Gen Meteor Lake
        - [ ] Novacustom V540TNx/V560TNx 14th Gen Meteor Lake
      - [ ] Community -> DPP (Coreboot + UEFI) with credentials -> Update should pass
        - [ ] MSI PRO Z790-P DDR4
        - [ ] MSI PRO Z790-P
        - [ ] MSI PRO Z690-A DDR4
        - [ ] MSI PRO Z690-A
        - [ ] Novacustom NV4x 11th Gen Tiger Lake
        - [ ] Novacustom NS5x/7x 11th Gen Tiger Lake 
        - [ ] Novacustom NS5x/7x 12th Gen Alder Lake
        - [ ] Novacustom NV4x 12th Gen Alder Lake
        - [ ] Novacustom V540TU/V560TU 14th Gen Meteor Lake
        - [ ] Novacustom V540TNx/V560TNx 14th Gen Meteor Lake

Notes

#### Restore
  - [ ] Restoring from cloud
  - [ ] Restoring from local storage
 
Notes:
* I do not remember if this feature was ever used by our team/clients. Is it working? Is it useful?

#### DPP Credentials test cases (platform independent):
  - [ ] Provide DPP Credentials with Dasharo firmware access and without DTS Extensions access -> DTS should have access to firmware only
  - [ ] Provide DPP Credentials without Dasharo firmware access and with DTS Extensions access -> DTS should have access to extensions only
  - [ ] Provide DPP Credentials with Dasharo firmware access and with DTS Extensions access -> DTS should have access to both products
  - [ ] Provide DPP Credentials without Dasharo firmware access and without DTS Extensions access -> DTS should not have access to any of the products
  - [ ] Replace already existing inside DTS credentials with new credentials -> the old credentials should be replaced without warnings/errors

Note: it is known that currently DPP credentials cannot be replaced without issues due to problems with `mc` (aka MinIO CLI) integration. This is to be fixed.

DTS Extensions test cases (platform independent):
- [ ] Install (DTS extensions are being installed during providing of DPP credentials) test DTS Extension
- [ ] Try to use the installed test DTS Extension

Note: this is the only functionalities for DTS Extensions right now.

pietrushnic, 2025-01-24:
s/Coreboot/coreboot/
s/Seabios/SeaBIOS/

Not sure what the conclusion is about h/Heads, but for now, I would follow: https://docs.dasharo.com/dasharo-naming-convention/#synopsis

m-iwanicki, 2025-07-08:
@DaniilKl Do we need to test `DPP -> Heads`, `Community -> Heads` separately? Is there even any difference? There are only couple of cases where we check fw version and do something different based on it e.g. [EC transition](https://github.com/Dasharo/dts-scripts/blob/bcf7871d4e334a47e7f8722793db5537db67a569/include/dts-functions.sh#L1491).

There is also  `Update UEFI Capsule DPP (coreboot + UEFI) -> DPP (coreboot + UEFI) ` case. I think we should check if DTS doesn't allow us to choose capsule update if our fw version is too low (doesn't support capsule). What if someone bought DPP subscription and wants to update and chooses capsule?

DaniilKl, 2025-07-09:
I have added transition from coreboot+SeaBIOS to coreboot+UEFI for PC Engines.

DaniilKl, 2025-07-09:
It is worth mentioning, that the UEFI Capsule Update, though supported by DTS is not used by any platforms currently. For the Heads update check this issue: https://github.com/dasharo/dasharo-issues/issues/1380.

DaniilKl, 2025-09-25:
After this PR https://github.com/Dasharo/open-source-firmware-validation/pull/949 we have all test cases from the list above:

<details><summary>The list</summary>
<p>

```bash
λ robot -L TRACE -v dts_config_ref:refs/heads/main -t "E2EH003.001*" dts/dts-e2e-helper.robot
==============================================================================
Dts-E2E-Helper                                                                
==============================================================================
E2EH003.001 Print names of test cases to be generated :: Print out... 
..msi-pro-z790-p-ddr5 Initial Deployment - DPP
msi-pro-z790-p-ddr5 UEFI Update - DPP
msi-pro-z790-p-ddr5 UEFI->Heads Transition - DPP
optiplex-7010 Initial Deployment - DPP
optiplex-7010 UEFI Update - DPP
optiplex-9010 Initial Deployment - DPP
optiplex-9010 UEFI Update - DPP
pcengines-apu2 UEFI Update - DPP
pcengines-apu2 SeaBIOS Update - DPP
pcengines-apu2 SeaBIOS->UEFI Transition - DPP
pcengines-apu3 UEFI Update - DPP
pcengines-apu3 SeaBIOS Update - DPP
pcengines-apu3 SeaBIOS->UEFI Transition - DPP
pcengines-apu4 UEFI Update - DPP
pcengines-apu4 SeaBIOS Update - DPP
pcengines-apu4 SeaBIOS->UEFI Transition - DPP
pcengines-apu6 UEFI Update - DPP
pcengines-apu6 SeaBIOS Update - DPP
pcengines-apu6 SeaBIOS->UEFI Transition - DPP
novacustom-nuc_box-125H Initial Deployment - DCR
novacustom-nuc_box-155H Initial Deployment - DCR
novacustom-v540tnd Initial Deployment - DCR
novacustom-v540tnd UEFI Update - DCR
novacustom-v540tnd Fuse Platform - DCR
novacustom-v540tu Initial Deployment - DCR
novacustom-v540tu UEFI Update - DCR
novacustom-v540tu UEFI->Heads Transition - DPP
novacustom-v540tu Fuse Platform - DCR
novacustom-v560tu Initial Deployment - DCR
novacustom-v560tu UEFI Update - DCR
novacustom-v560tu UEFI->Heads Transition - DPP
novacustom-v560tu Fuse Platform - DCR
msi-pro-z690-a-ddr5 Initial Deployment - DCR
msi-pro-z690-a-ddr5 Initial Deployment - DPP
msi-pro-z690-a-ddr5 UEFI Update - DCR
msi-pro-z690-a-ddr5 UEFI Update - DPP
msi-pro-z690-a-ddr5 UEFI->Heads Transition - DPP
msi-pro-z690-a-wifi-ddr4 Initial Deployment - DCR
msi-pro-z690-a-wifi-ddr4 Initial Deployment - DPP
msi-pro-z690-a-wifi-ddr4 UEFI Update - DCR
msi-pro-z690-a-wifi-ddr4 UEFI Update - DPP
msi-pro-z690-a-wifi-ddr4 UEFI->Heads Transition - DPP
novacustom-ns50mu Initial Deployment - DCR
novacustom-ns50mu UEFI Update - DCR
novacustom-ns50pu Initial Deployment - DCR
novacustom-ns50pu UEFI Update - DCR
novacustom-ns70mu Initial Deployment - DCR
novacustom-ns70mu UEFI Update - DCR
novacustom-ns70pu Initial Deployment - DCR
novacustom-ns70pu UEFI Update - DCR
novacustom-nv41mb Initial Deployment - DCR
novacustom-nv41mb UEFI Update - DCR
novacustom-nv41mz Initial Deployment - DCR
novacustom-nv41mz UEFI Update - DCR
novacustom-nv41pz Initial Deployment - DCR
novacustom-nv41pz UEFI Update - DCR
novacustom-nv41pz UEFI->Heads Transition - DPP
novacustom-v560tnd Initial Deployment - DCR
novacustom-v560tnd UEFI Update - DCR
novacustom-v560tnd Fuse Platform - DCR
novacustom-v560tne Initial Deployment - DCR
novacustom-v560tne UEFI Update - DCR
novacustom-v560tne Fuse Platform - DCR
odroid-h4-plus Initial Deployment - DPP
odroid-h4-plus UEFI Update - DPP
odroid-h4-plus Dasharo (coreboot+UEFI) to Dasharo (Slim Bootloader+UEFI) Transition - DPP
odroid-h4-plus Dasharo (Slim Bootloader+UEFI) Initial Deployment - DPP
E2EH003.001 Print names of test cases to be generated :: Print out... | PASS |
------------------------------------------------------------------------------
Dts-E2E-Helper                                                        | PASS |
1 test, 1 passed, 0 failed
==============================================================================
λ grep -rn 'E2E[0-9][0-9][0-9].[0-9][0-9][0-9]' dts/dts-e2e.robot
34:E2E001.001 HCL Report test
56:E2E002.001 DCR Initial Deployment On Msi-pro-z690-a-ddr5 With 13th Gen CPU Should Fail
61:E2E002.002 DCR Initial Deployment On Msi-pro-z690-a-wifi-ddr4 With 13th Gen CPU Should Fail
66:E2E003.001 DCR UEFI Update On Msi-pro-z690-a-ddr5 With 13th Gen CPU Should Fail
71:E2E003.002 DCR UEFI Update On Msi-pro-z690-a-wifi-ddr4 With 13th Gen CPU Should Fail
80:E2E007.001 Check credentials are being saved correctly
101:E2E007.002 Check old credentials are being overwritten by new
129:E2E007.003 Check wrong credentials should not allow to log into DPP services
149:E2E007.004 Check correct credentials should allow to log into DPP services
169:E2E007.005 Check empty e-mail should not pass
184:E2E007.006 Check empty password should not pass
200:E2E007.008 Check DPP credentials with access to only firmware
219:E2E007.009 Check DPP credentials with access to only extensions
238:E2E007.010 Check DPP credentials without DPP access
259:E2E007.011 Check DPP credentials with both DPP firmware and DTS extensions access
274:E2E008.001 Reboot UI Option Calls Reboot Command
283:E2E008.002 Poweroff UI Option Calls Poweroff Command
292:E2E008.003 Launch SSH Server UI Option Enables SSH Server Command
306:E2E008.004 Enable Sending Logs UI Option Should Enable DTS Log Sending
320:E2E009.001 DTS extensions are installed and can be used
```

</p>
</details> 

Now we should focus on:
* Adding clear and concise definition of success/error paths.
* Automate/add process on how to add the tests, ensuring it can be added with close to zero effort.

closed 2025-09-25 by DaniilKl