Dasharo/open-source-firmware-validation #666: Ability to use fully automated tests on all platforms

DaniilKl, 2025-01-15:
### The problem you're addressing (if any)

Our tests mostly use serial interface and for some platforms it is difficult to get access to serial (NovaCustom platforms for example). Therefore, we have to do some complex steps to prepare for testing, like disassembling a platform, soldering wires to UART etc..

### Describe the solution you'd like

Maybe we could find another way to get access to Firmware/Linux shell, e.g. redirecting the shells to USB port and then reading it directly via USB or via UART using USB/UART converter connected to RTE?

### Where is the value to a user, and who might that user be?

Fully automated, both DST and Dasharo tests an all platforms, reducing developing and release time.

DaniilKl, 2025-01-15:
@m-iwanicki, @filipleple.

filipleple, 2025-01-15:
@macpijan I think the idea here is not obtaining early coreboot console logs, but only being able to navigate the EDK2 menus to switch options - which is after the USB stack has been initialized, so I'm guessing we could come up with an efi driver that would bridge the console to one of the USB ports.

The question remains if this would really help the testing so much that it would be worth the effort. It would probably cut down  time spent maintaining `dcu` and the osfv library for it, and the extra effort spent setting up a laptop testing stand

DaniilKl, 2025-01-22:
@macpijan, so, considering @filipleple and @mkopec answers, wouldn't it be worthy to try to use USB2UART inside firmware for firmware menu navigation? It will definitely help for testing DTS, and IIUC Dasharo firmware too.

macpijan, 2025-01-22:
Yes always worth to explore new ideas as we come up with them.

macpijan, 2025-01-22:
> The question remains if this would really help the testing so much that it would be worth the effort. It

I think it is, we have done even more in the past.

filipleple, 2025-04-16:
Implemented in 

https://github.com/Dasharo/coreboot/pull/654
https://github.com/Dasharo/edk2/pull/239
https://github.com/Dasharo/edk2-platforms/pull/1

closed 2025-04-16 by filipleple