Dasharo/open-source-firmware-validation #668: Adding time synchronization to network boot tests

mysteriousCzrsX, 2025-01-16:
### The problem you're addressing (if any)

Network boot tests PXE004.001 DTS option is available and works correctly and PXE005.001 OS installation option is available and works correctly. Require that the system boots over IPXE using https. When the system time is not synchronized, the https connection would fail, thus the tests would not be completed. 

### Describe the solution you'd like

We could synchronize time prior to running the tests by using `ntp` command in iPXE shell. Command is described [here](https://ipxe.org/cmd/ntp). 

### Where is the value to a user, and who might that user be?

This would save time when running test since now to properly run the tests a manual time synchronization procedure must be performed.

### Describe alternatives you've considered

An alternative method of time synchronization is booting in to the operating system and using its tools to synchronize the system clock.

SebastianCzapla, 2025-02-04:
[image]
Many header and Suite inits are mess of multiple keywords and formatting. Perhaps this suite is another one that could benefit greatly from isolating initialization into local keyword. We could also force network sync in the local init keyword.

olety, 2026-09-08:
Hi, I reviewed PR #1268. Its synchronization helper logs the `ntp` output without checking success. I could add an explicit success check, a useful failure message, and regression coverage, in coordination with the existing PR author.

Would that follow-up be useful and eligible for any of this issue's stated $40 bounty? Would validation of PXE004/PXE005 using the documented Dasharo QEMU Q35 setup be accepted, or is a particular hardware setup required?

This is a source-level finding; I have not run the patch.

Th0masclassic, 2026-09-08:
Submitted replacement PR #1283 for the stated $40 bounty. It keeps the May approach scoped to PXE004/PXE005, but additionally verifies the iPXE `ntp` command status before HTTPS boot and fails with the captured NTP output if synchronization did not succeed. Robot parser and the repo-pinned Robocop 6.5.0 lint/format checks pass locally. Hardware execution was not claimed.