Dasharo/open-source-firmware-validation #675: Platform config sharing fails in some cases

macpijan, 2025-01-16:
In RF, we cannot override variable by defining it again.

Value which was first defined is used.

This makes the platform config scheme prone to simple errors.

 Example

1. We set `${DEVICE_AUDIO1}=                                   ${TBD}` in `default.robot`
2. We set another value in `protectli-vp2410` which imports `default.robot`.
3. The resulting value is `${TBD}`, not what we set in `protectli-vp2410`, as the values from imported resources have been parsed first.

This might have not been spotted before, as most of the platforms use further common configs (such as `protectli-common.robot`, not including `default.robot` directly.

philipandag, 2025-02-03:
It should always be possible to overwrite a variable in a file. What could be tricky is making sure the variables are properly overwritten by the resource files.
I think we should just take care not to import anything twice or remember to do it in correct order.

And if that is not enough, then an import order can be defined, and dynamically changed to allow overwriting: https://github.com/Dasharo/open-source-firmware-validation/pull/637/files#diff-2bd69584d51e03632b6b478cae660b9d98acb614bedfcf8ad005f876c9d6744eR503

I think we could even dynamically generate a list of all libraries and enforce a specific order by running "Set Library Search Order" with a crazy long list of arguments :p

wiktormowinski, 2025-04-09:
As of now i was unable to stumble upon such case. I'll include a short guide on how to prevent such issue (maybe this will help in the future since it's easier to link infoto someone in need): https://github.com/Dasharo/open-source-firmware-validation/blob/9f7fcbb32c6ab1cd943017da87933717192bcb14/docs/contributing.md#variable-import-rules