Dasharo/open-source-firmware-validation #680: Refactor QEMU lib

macpijan, 2025-01-21:
Problems:

Errors from QemuMonitor are ignore, so many problems are ignored under the hood. We do now know when things fail in QemuMonitor. That was the result of false positive in some tests in the past, it seems.

Catching errors has been added here: b956e6fb9d60

But it results in stockpiling errors when handling disk / USB connection / disconnection.

A good suite to test this is `esp-scanning`.

The end result of this task should be that the esp-scanning and secure-boot suite can reliably pass on QEMU.

macpijan, 2025-01-21:
Some experiments done here: https://github.com/Dasharo/open-source-firmware-validation/tree/qemu-lib-refactor

The results led me to believe that the current implementation has never worked correctly, so I have up at this point.

One could start from checking the state of this suite prior any mods.

philipanda, 2025-04-14:
The qemu host does not detect any hotplugged disks. The esp-scanning disk can be attached as an ide-hd, which only supports plugging it using args to qemu-system. Connecting an SCSI drive the same way does not work. Possibly related to https://github.com/Dasharo/dasharo-issues/issues/901