Dasharo/open-source-firmware-validation #683: TPM002.001 test case false positive

wiktormowinski, 2025-01-23:
### Device

VP6650

### RTE version

-

### OSFV version

develop, https://github.com/Dasharo/open-source-firmware-validation/commit/7df966b73cb356f02f3909b23ac9a8275003d3fe

### Affected component(s) or functionality

TPM002.001 returned PASS when the TPM module is not even plugged in

### Brief summary

TPM002.001 returned PASS when the TPM module is not even plugged in

### How reproducible

100%

### How to reproduce

run this test without TPM module

### Expected behavior

it should not detect anything when tpm is physically not there

### Actual behavior

it passes the tpm detection

### Link to screenshots or logs

[dasharo-security_log.zip](https://github.com/user-attachments/files/18521115/dasharo-security_log.zip)

philipandag, 2025-02-03:
Won't adding `-1` to the cbmem command help?

If CBMEM base address is always the same, (is it?) then wouldn't that just mean that the TPM log is not overwritten when there is no TPM installed?

If that's the case then we could say that TPM001.001 and TPM002.001 should only be considered as a PASS if TPM003.001 also passes.

SebastianCzapla, 2025-02-04:
Fixed in https://github.com/Dasharo/open-source-firmware-validation/pull/507

closed 2025-02-04 by SebastianCzapla