Dasharo/open-source-firmware-validation #696: OSFV for systems without UEFI Shell + Netboot autoboot update

SebastianCzapla, 2025-02-10:
### The problem you're addressing (if any)

Currently, OSFV relies on UEFI shell for variety of keywords and tests. Systems where UEFI shell is disabled require additional work for OSFV compatibility.

### Describe the solution you'd like

Refactor tests and keywords, to include guards to resolve UEFI Shell dependency and skip/warn properly.
Update default configs for affected platforms.

Update uefi-shell.robot to test that expected state of UEFI Shell is there, if it is disabled in config, it must not appear in the boot options.
USH001.001 UEFI Shell 
Do not skip this test on lack of UEFI_SHELL_SUPPORT, instead verify that no "UEFI Shell" appears in the menu constructions.

Update network-boot.robot to test scenario where autoboot is disabled. Update config with flag that conveys expected state of autoboot, and run either PXE003.001 or new test:
PXE008.001 Autoboot option is properly disabled
To detect and verify no autoboot, we first verify that autoboot string does not appear in submenu construction, and then we can wait 60-120 seconds and read for expected output, to verify that we did not boot during last minute.

### Additional context

List of tests that use UEFI Shell and may need flag addition:
USB002.001 USB keyboard detected in FW

miczyg1, 2025-02-10:
> Update network-boot.robot to test scenario where autoboot is disabled. Update config with flag that conveys expected state of autoboot, and run either PXE003.001 or new test:
PXE008.001 Autoboot option is properly disabled
> To detect and verify no autoboot, we first verify that autoboot string does not appear in submenu construction, and then we can wait 60-120 seconds and read for expected output, to verify that we did not boot during last minute.

Autoboot word is not present in Protectli iPXE menus. It will be just enough to read the menu completely and after 15 seconds check if any action was taken (there shouldn't;t be any). Regualr Dasahro iPXE menu has not been modified. So it shall affect only NBU tests (https://github.com/Dasharo/open-source-firmware-validation/blob/main/dasharo-compatibility/network-boot-utilities.robot). Tests from PXE suite (https://github.com/Dasharo/open-source-firmware-validation/blob/main/dasharo-compatibility/network-boot.robot) should stay intact.

> USB002.001 USB keyboard detected in FW

This will be a bit complicated. To be honest, I currently have no idea how to test that without the shell. Maybe the Boot Maintenance Manager will list USB keyboard if present (https://docs.dasharo.com/dasharo-menu-docs/boot-maintenance-mgr/#console-options) which may be an indicator of working keyboard. But this needs verification.

m-iwanicki, 2025-02-10:
Maybe replace built-in UEFI shell with external EFI shell booted via `efi` file or network boot? I think it should work at least in tests that don't test UEFI shell itself (or SB related ones)

macpijan, 2025-02-10:
For testing with SB disabled, we can load shell via network.

For SB tests that specifically need SB to be enabled, we can rewrite them to not execute efi files from shell, and use boot manager instead.

SebastianCzapla, 2025-02-13:
https://github.com/Dasharo/open-source-firmware-validation/pull/697 adds bare minimum to test the new UEFI Shell defualts. Should we separate refactoring of SB tests into different PR or keep all the changes together?

SebastianCzapla, 2025-02-14:
@macpijan Let me know if https://github.com/Dasharo/open-source-firmware-validation/pull/697 is enough for the initial PR

macpijan, 2025-02-14:
To me it is, if someone helse has tested it

SebastianCzapla, 2025-02-14:
@macpijan Here's an archive with runs for both tests, on rc8 and rc6 of vp3230 platform [no-uefi-shell-logs.zip](https://github.com/user-attachments/files/18798173/no-uefi-shell-logs.zip)

macpijan, 2025-03-05:
@BeataZdunczyk This should stay open, linked MR solves it only partially. Please see the original post.

BeataZdunczyk, 2025-04-08:
@macpijan , @SebastianCzapla Can we close this issue now? This PR has been merged: https://github.com/Dasharo/open-source-firmware-validation/pull/725

SebastianCzapla, 2025-04-17:
Closing, as the goals of this issue have been achieved, and relevant PRs have been merged

closed 2025-04-17 by SebastianCzapla