Dasharo/open-source-firmware-validation #702: DCU004.001 Verify SMMSTORE changes fails on file transfer

SebastianCzapla, 2025-02-24:
### Device

vp3230

### RTE version

--

### OSFV version

develop (with small tweaks that should not impact the test)

### Affected component(s) or functionality

DCU004.001 Verify SMMSTORE changes

### Brief summary

Test fails to execute

### How to reproduce

Run the DCU004.001 test on the vp3230

### Expected behavior

Test runs and verifies correct behavior

### Actual behavior

Test fails on file transfer step, with 104 error (Connection closed by peer) or with coreboot.rom file not found error.

### Link to screenshots or logs

[dcu.zip](https://github.com/user-attachments/files/18940853/dcu.zip)

BeataZdunczyk, 2025-03-28:
@SebastianCzapla did this issue reoccur with the recent rc?

SebastianCzapla, 2025-04-17:
Closing as issue is tracked here https://github.com/Dasharo/open-source-firmware-validation/issues/771 and has more details.

closed 2025-04-17 by SebastianCzapla