Dasharo/open-source-firmware-validation #714: osfv robot (orobot) checked in & checked out DUT during nightly test run

matmacieje, 2025-03-07:
### Device

vp2420

### RTE version

not relevant

### OSFV version

develop

### Affected component(s) or functionality

dasharo-performance/platform-stability.robot

### Brief summary

DUT chcekd out and kept by osfv robot (orobot) , nightly test run failed

### How reproducible

reproduction 1/2, 50%

### How to reproduce

Run some long test suite over night on checked out DUT, wait.

### Expected behavior

Let me finish testing, while DUT is checked out on me in SnipeIT.

### Actual behavior

Interrupted test run.

### Link to screenshots or logs

[20250306_08.52.11.zip](https://github.com/user-attachments/files/19126031/20250306_08.52.11.zip)

### Solutions you've tried

Re-run following night.

BeataZdunczyk, 2025-03-12:
@macpijan Do you know what could be the cause of this issue? Maybe some missing configuration?

@JakubRed had the same problem, but only the two of them encountered this issue.

macpijan, 2025-03-12:
I think I do.

Checkout/checkin happens for each suite (each .robot file) once https://github.com/Dasharo/open-source-firmware-validation/blob/main/keywords.robot#L268

If we run multiple suites one by one, we have multiple checkouts / checkins. So if it happens that osfv-robot check was in between, it could reserve platform for itself.

The solution for this right now is to checkout the platform manually before starting regression. This way it should not checkin the platform as shown in the test.

BTW, the osfv-robot check has been disabled right now anyway, as it failed to checking platforms.

JakubRed, 2025-03-12:
> @JakubRed had the same problem, but only the two of them encountered this issue.

Actually, I was @matmacieje that encountered this during the night tests.

matmacieje, 2025-04-15:
No further reproduction.

closed 2025-04-15 by matmacieje