Dasharo/open-source-firmware-validation #716: WLE test in minimal-regression fail

miczyg1, 2025-03-11:
### Device

VP66xx

### RTE version

1.1.0

### OSFV version

e8abd7a3d2bd185a22db12ebafda018fddc8ded7

### Affected component(s) or functionality

WLE002.001

### Brief summary

The test does nto check if package providing nmcli is installled

### How reproducible

Always

### How to reproduce

Run:

`FW_FILE=protectli_vp66xx_v0.9.1.rom DEVICE_IP=192.168.10.132 RTE_IP=192.168.10.110 CONFIG=protectli-vp6670 ./scripts/regression.sh -- --include "minimal-regression"`

### Expected behavior

WLE002.001 passes

### Actual behavior

```
==============================================================================
Dasharo-Compatibility.Wifi-Bluetooth-Support                                  
==============================================================================
WLE002.001 Wi-Fi scanning (Ubuntu) :: Check whether the Wi-Fi func... ..Remember to test all variants of wireless cards.
[ WARN ] Remember to test all variants of wireless cards.                     
WLE002.001 Wi-Fi scanning (Ubuntu) :: Check whether the Wi-Fi func... | FAIL |
No match found for '3mdeb_Laboratorium' in 1 minute. Output:
snap install network-manager  # version 1.2.2-31, or
apt  install network-manager  # version 1.44.2-1ubuntu2
See 'snap info network-manager' for additional versions.
nmcli device wifi list
root@3mdeb:/home/ubuntu# nmcli device wifi rescan
Command 'nmcli' not found, but can be installed with:
snap install network-manager  # version 1.2.2-31, or
apt  install network-manager  # version 1.44.2-1ubuntu2
See 'snap info network-manager' for additional versions.
root@3mdeb:/home/ubuntu# nmcli device wifi list
Command 'nmcli' not found, but can be installed with:
snap install network-manager  # version 1.2.2-31, or
apt  install network-manager  # version 1.44.2-1ubuntu2
See 'snap info network-manager' for additional versions.
root@3mdeb:/home/ubuntu#

```

### Link to screenshots or logs

See logs above

matmacieje, 2025-03-28:
@miczyg1 could you provide Ubuntu version? I made tests on Ubuntu 24.04 LTS, I'm assuming that this is version we're moving to. network-manager package (provider of nmcli) is installed there by default, purging it requires manual interface configuration after reboot. Besides, I've learned that there will be Ansible-based dependency management mechanism (incl. package names translation) deployed very soon in OSFV and local package management in test suites is highly discouraged. Do you think that simple check of command existence (and test SKIP, in case of failure) would be enough fix?

miczyg1, 2025-03-28:
> [@miczyg1](https://github.com/miczyg1) could you provide Ubuntu version? I made tests on Ubuntu 24.04 LTS, I'm assuming that this is version we're moving to. network-manager package (provider of [...]

Ubuntu version doesn't matter. If the test requires `nmcli` it should ensure it is installed, and if not, install it. Not skip the test... Tests must be robust

BeataZdunczyk, 2025-03-28:
Please do not close the issue until the PR is merged and properly tested @matmacieje

closed 2025-04-15 by BeataZdunczyk