Dasharo/open-source-firmware-validation #722: Multiple Login To Windows keywords conflicting silently via SSH

SebastianCzapla, 2025-03-13:
### Device

All Novacustom (SSH Dut Conenection specific)

### RTE version

None

### OSFV version

Develop

### Affected component(s) or functionality

Automated testing of Novacustom devices for Windows OS

### Brief summary

Tests that need Login To Windows usign SSH as DUT connection method, will default to DCU implementation of this Keyword, which fails on laptops, due to fixed bootloader order.

### How reproducible

100%

### How to reproduce

Setup Novacustom device for testing.
Select Windows as first option in boot order menu.
Setup OSFV and run any test that requires Login To Windows, for example dasharo-compatibility/cpu-status.robot

### Expected behavior

RF Logins to Windows, tests continue

### Actual behavior

RF attempts to setup Windows as boot priority, and does so by first logging into Ubuntu.

### Link to screenshots or logs

[cpu-status.zip](https://github.com/user-attachments/files/19227859/cpu-status.zip)

### Solutions you've tried

Possible workaround is to use keywords.Login To Windows

philipanda, 2025-03-13:
The PR https://github.com/Dasharo/open-source-firmware-validation/pull/723 does not fix it completely.
The fact that `Power On` does nothing on laptops has to be either changed or taken into account in some way in every test suite, because if Power On does nothing, then Windows is not shut down after a test case ends.

Laptop tests assume, that some kind of Linux is booted initially, but that is not the case after running a Windows test case, which won't boot back a Linux.

SebastianCzapla, 2025-04-09:
Explanation of the issue here https://github.com/Dasharo/open-source-firmware-validation/pull/723

Closing as this is not a OSFV Issue, rather a new, previously undocumented behavior.

closed 2025-04-09 by SebastianCzapla