Dasharo/open-source-firmware-validation #726: Add fedora variants of Ubuntu test cases

philipanda, 2025-03-13:
According to the naming convention: https://github.com/Dasharo/open-source-firmware-validation/blob/develop/docs/tests-naming-convention.md

Most of a test case logic can be moved to a keyword only leaving a test case ID and required SKIPS, like in https://github.com/Dasharo/open-source-firmware-validation/blob/develop/dasharo-compatibility/display-ports-and-lcd-support.robot#L46-L73

The test cases to implement were separated into a few groups, where implementing the test cases in a given group could cause similar problems.

Most, if not all of them, should probably work out of the box if only the correct distribution is booted.

The list of the test cases is based on the V54xNx_MTL V54xNx_MTL v0.9.1-rc8 test scope from https://docs.google.com/spreadsheets/d/1wSE6xA3K3nXewwLn5lV39_2wZL1kg5AkGb4mvmG3bwE/edit?gid=382389731#gid=382389731

### Should just work

- TPM,TPD
  - sysfs and cbmem
  - should work
- MBO
  - reads PCRS, sysfs
- MNE
  - checking ME, sysfs
  - should work
- USB
  - lsusb,
  - should just work
- NVM
  - lspci,
  - missing TESTS_IN_FW_SUPPORT
  - should work
- WLE
  - pciutils, lspci, bluetoothctl,
  - mutliple packages
  - should work
- CBMEM
  - rtcwake, cbmem
  - should work if cbmem works on fedora
- DSP
  - sysfs,
  - should work
- AUD
  - alsa-utils package, sysfs,
  - should work  
- MWL
  - lspci, nmcli, bluetoothctl
  - should work
- DMI
  - dmidecode
  - should work
- NET
  - fwts, rtcwake, ip
  - should work, but different kernels
    might name the interfaces differently
- DCU
  - dcu, docker
  - should just work
- STB
  - /proc/uptime, ip
  - should just work
- CPU
  - getconf, lscpu
  - should just work
- TPMCMD
  - tpm2-tools - package available
  - should just work
- SDC
  - lsblk
  - should just work

### Reboots, warmboots - testing might take long

- SUD
  - lsusb, rtcwake, fwts, packages available,
  - should work
- SNV
  - lspci, rtcwake, fwts
  - should work
- SUSP
  - only fwts
  - should just work
- UTC
  - fwts
  - should work
- SMW
  - lspci, pciutils, rtcwake, fwts, nmcli
  - Boot Operating System
    - duplicate of `Boot System Or From...`?
    - try replacing it, because only the
    latter is able to boot selected OS
    on laptops
  - lspci, rtcwake
  - otherwise should work
- SUD
  - lsusb, fwts, rtcwake
  - reboots, warmboots
  - should work
- SNV
  - lspci, fwts, rtcwake
  - the same as SUD but lspci
  - should work

### Missing/different name packages

- CAM
  - libinput packages different names,
  - ffmpeg
  - might cause some problems
- ECR
  - acpi-call, modprobe, lots of sysfs,
  - libinput-tools - wrong package name
- NVI
  - pciutils, mesa-utils - no package,
  - sysfs

### Keyword renaming and little tweaks

- CPT - lib/sensors
  - not updated to
  the new lib/sensors.robot
  SMW  - `Prepare Sensors` should be added to suite
      setup
    - manually perparing lm-sensors should be
      removed
    - might require adding a config based on
      platform-configs/include/sensors/default-sensors-config.yaml
      , should work then
  - change keyword names (Ubuntu) if suitable
- CPF
  - some ugly workaround for booting in HEADS,
  should be implemented into Boot System...
  instead

### Enable/disable, rewrite to the naming convention

- BLS
  - install flashrom,
  - enable/disable test, needs rewriting
- WBS
  - lspci, lsusb
  - rename keywords containing (Ubuntu)
  - enable/disable test, needs rewriting
- CHS
  - lsusb
  - should just work
  - enable/disable test, needs rewriting

BeataZdunczyk, 2025-03-14:
> The list of the test cases is based on the V54xNx_MTL V54xNx_MTL v0.9.1-rc8 test scope from https://docs.google.com/spreadsheets/d/1wSE6xA3K3nXewwLn5lV39_2wZL1kg5AkGb4mvmG3bwE/edit?gid=382389731#gid=382389731

Please check the test matrix in our test matrix tool, where we have the most recent version for NovaCustom releases.
For reference, you can also find the published test results here: https://github.com/Dasharo/osfv-results/tree/main/boards/NovaCustom/MTL_14th_Gen

philipanda, 2025-04-04:
PRs that add the test cases:

https://github.com/Dasharo/open-source-firmware-validation/pull/747
https://github.com/Dasharo/open-source-firmware-validation/pull/749
https://github.com/Dasharo/open-source-firmware-validation/pull/753
https://github.com/Dasharo/open-source-firmware-validation/pull/761

closed 2025-07-01 by BeataZdunczyk