Dasharo/open-source-firmware-validation #731: Permission Denied during Get Cbmem KWD

SebastianCzapla, 2025-03-18:
### Device

--

### RTE version

--

### OSFV version

develop

### Affected component(s) or functionality

Test that require cbmem

### Brief summary

PermissionError: [Errno 13] Permission denied

### How reproducible

100%

### How to reproduce

Run dasharo-performance/boot-time-measure.robot on Novacustom device over SSH

### Expected behavior

cbmem is downloaded, tests continue

### Actual behavior

Tests fail on the Send file to DUT Keyword

### Link to screenshots or logs

[dasharo-performance.zip](https://github.com/user-attachments/files/19317540/dasharo-performance.zip)

miczyg1, 2025-03-19:
The SSH session is made on a user ubuntu not root, that is why the permission error occurs maybe?

SebastianCzapla, 2025-04-02:
Fixed in: https://github.com/Dasharo/open-source-firmware-validation/pull/754

SebastianCzapla, 2025-04-29:
Fix merged, closing

closed 2025-04-29 by SebastianCzapla