Dasharo/open-source-firmware-validation #734: Out of bound access to config list

miczyg1, 2025-03-19:
### Device

VP3210

### RTE version

Any

### OSFV version

develop

### Affected component(s) or functionality

`Get Current CONFIG Stop Index` keyword

### Brief summary

robot is unable to find the config if given config is the last in the list

### How reproducible

Always

### How to reproduce

`robot -L TRACE -v rte_ip:192.168.10.211  -v config:protectli-vp3210 -v device_ip:192.168.10.21 -v fw_file:protectli_vp32xx_v0.9.0-rc10.rom dasharo-compatibility/usb-hid-and-msc-support.robot`

### Expected behavior

Suite setup doesn't fail.

### Actual behavior

Suite setup fails.

### Link to screenshots or logs

Logs attached

[log.zip](https://github.com/user-attachments/files/19342696/log.zip)

### Additional context

Problem is in `Get Current CONFIG Stop Index` keyword. The line
`${index}=    Set Variable    ${start+1}`  causes the robot to not enter the FOR loop, because `${index}`is incremented to 122 (the exact size of `${config_list}`). So it will always fail to find config for the last platform in the `${config_list}`, whatever it would be.

miczyg1, 2025-03-20:
@macpijan the faulting line `${index}=    Set Variable    ${start+1}` is in the keywords.robot since the initial commit. SO I am not sure who was the author and what was the intent here. Can you help here?

wiktormowinski, 2025-04-15:
@miczyg1 looks like we got rid of this keyword entirely:

[image]
_Fig 1. Fragment of log from current `develop`, issue not prevalent. Reason: problematic line removed._ 

Full log attached:

[no_reproduction.zip](https://github.com/user-attachments/files/19761195/no_reproduction.zip)

wiktormowinski, 2025-04-15:
i don't know if i understand it correctly but this discovery implies the `variables.robot` file is not used in the current osfv workflow???

miczyg1, 2025-04-16:
The keywords are still used. Just grepping `Get Current CONFIG` is enough. Although it is not used in Suite Setup anymore, it is still used by `Boot System Or From Connected Disk`. So IMO it may still fail.

> i don't know if i understand it correctly but this discovery implies the variables.robot file is not used in the current osfv workflow???

Why? It is imported as a resource in many files.

miczyg1, 2025-04-16:
It doesn't reproduce on current develop (18056eaf6749dfd3969723c2bee06702214f5c86), but it still reproduces on 929119d59f00a98bbc8da873b06cc6526311f54f (where I originally found this issue).

I can check what has changed that could potentially fix or hide the problem...

wiktormowinski, 2025-04-16:
it looks like it can be triggered when platform has no boot entries that contain os name:

[image]
in this case this whole condition got skipped

miczyg1, 2025-04-16:
[log.zip](https://github.com/user-attachments/files/19775681/log.zip)

closed 2025-04-16 by miczyg1