Dasharo/open-source-firmware-validation #738: Change DTS tests to conform to new DPP credentials

PLangowski, 2025-03-21:
### The problem you're addressing (if any)

After migrating to MinIO, we will have one set of credentials for all platforms:
- Email
- Password

DTS e2e tests need to be changed accordingly.

m-iwanicki, 2025-06-05:
@PLangowski I think this issue can be closed: https://github.com/Dasharo/open-source-firmware-validation/blob/c26052acf70c3f79c093821ec1645e67f7613e40/lib/dts-lib.robot#L203-L215

DaniilKl, 2025-07-03:
I agree with @m-iwanicki.