Dasharo/open-source-firmware-validation #748: Reduce unnecessary reboots in tests

philipanda, 2025-03-29:
### The problem you're addressing (if any)

Tests that are performed on multiple OSes generally perform a lot of unnecesary reboots. Instead of running all little checks on one OS and the moving to another, we are doing a little check on every supporteed OS before moving to another one.

Example: the DSP test suite. The DSP001, DSP002, DSP003 test on one OS can be done in one go. The total execution time would be very close to executing just one of them, because most of the time is wasted on rebooting the platform and the OS. 

### Describe the solution you'd like

The tests should not perform reboots if it is not necessary.
How to solve this is not straightforward.

### Where is the value to a user, and who might that user be?

A significant decrease of tests run time, especially on regressions.

### Describe alternatives you've considered

Ideas

1. The order of executing the tests has to be changed
  - Just swap the test cases in test suites, so that all cases on one OS are next to each other. RF executes cases in
    the order they appear in the .robot file
  - separate tests on different OSes to their own files
    - could make it hard to run one test suite on all supported environments
      - the test file name when running robot can include a wildcard character, which might make this a good solution
      - `scripts/run.sh dasharo-compatibility/network*.robot` works and runs `network-boot-utilities`, `network-boot` and `network-speed` in one run.
2. After the order is altered, we need to modify the tests to not reboot if unnecessary
  - Keeping the state of the platform and making the `Power On` keyword work more like
    `Ensure Power On`.
    - many tests use `Power On` as a way to reboot the device. Such cases would need to be fixed. Maybe a
      separate keyword `Power Off` could be added and used to clearly state when the platform is actually
      rebooted, and when the `Power On` is used only to ensure it is On.
  - If the tests on different OSes would be separated to their own files, they could be simplified, so that Powering
   On the platform only happens during test suite setup, not on test case basis.
  - If tests on different OSes would stay in one file, a dummy test case could be added before OS change, that 
    would perform the reboot and boot the OS used in the following tests. 
    - definitely not a pretty solution, could be easy to implement, not so much to use later

### Additional context

https://github.com/Dasharo/open-source-firmware-validation/issues/686
https://github.com/Dasharo/open-source-firmware-validation/issues/652

krystian-hebel, 2025-03-29:
One way of making them faster would be to gather all the required info in suite setup and just parse it in dedicated test, like I did in https://github.com/Dasharo/open-source-firmware-validation/pull/559. This of course depends on exact tests, SMBIOS was easy to do in one go, but others may require multiple boots anyway.

> separate tests on different OSes to their own files

This would probably be required if the solution I proposed were to be used.

> the test setup could also just contain a `Skip` if the OS, for which tests this file contains, is not set as supported in
config variables

The problem with that approach would be that with each change to scope of tests in the release, we would have to update platform config. I like the `-t *.201` idea better, or we can [use tags](https://robotframework.org/robotframework/latest/RobotFrameworkUserGuide.html#by-tag-names) for this.

philipanda, 2025-04-17:
> One way of making them faster would be to gather all the required info in suite setup and just parse it in dedicated test, like I did in [...]

A way to achieve this would be to always run tests on only one environment, like one operating system. Switching between them automatically could be achieved by extending the `scripts/run.sh` script so that it would run the tests multiple times and change the ID of the OS that will be tested. The results could then be fused together using `rebot`. This way pretty much every test suite could be written in the way proposed by you in #559

philipanda, 2025-09-18:
I'd like to emphasize how much of a difference reducing the wasteful reboots makes.
Here are some screenshots from Krystian's PR he linked in his comment:
> Before:
> [image]
> 
> After:
> [image]

philipanda, 2025-11-14:
A solution to this issue that remembers the state of the platform and performs reboots automatically if needed was closed as it would be a bit too complex to be comfortably used now https://github.com/Dasharo/open-source-firmware-validation/pull/1141

We've decided to settle on a much simpler way to achieve similar results, that on the other hand requires more work on each test suite. Instead of remembering the state of the platform, the execution can be planned more smartly by splitting the suites into multiple files: https://github.com/Dasharo/open-source-firmware-validation/pull/1150

The issue shouldn't get closed yet, as it still applies to most test suites in OSFV. Each one needs to be changed in similar way as AUD in the PR above to reduce the reboots significantly