Dasharo/open-source-firmware-validation #764: Not existing snipeit variables in test suite teardown

JakubRed, 2025-04-07:
### Device

v540tu

### RTE version

none

### OSFV version

develop

### Affected component(s) or functionality

cpu-status.robot, cpu-frequency.robot and cpu-temperature.robot - test suites

### Brief summary

The listed test suites fail becouse of ${SNIPEIT_ALREADY_CHECKED_OUT_MANUALLY}' not found. In test suite teardown.

### How reproducible

100%

### How to reproduce

Run any of the above test suites.

### Expected behavior

Teardown running correctly

### Actual behavior

FAIL: Evaluating IF condition failed: Variable '${SNIPEIT_ALREADY_CHECKED_OUT_MANUALLY}' not found.

### Link to screenshots or logs

[CPU_TeardownFail.zip](https://github.com/user-attachments/files/19632269/CPU_TeardownFail.zip)

[CPT.zip](https://github.com/user-attachments/files/19632275/CPT.zip)

[CPF_TeardownFail.zip](https://github.com/user-attachments/files/19632290/CPF_TeardownFail.zip)

SebastianCzapla, 2025-04-07:
This variable is created as part of "Open Connection And Log In" which is invoked during "Prepare to Serial Connection" but is not a part of "Prepare To SSH Connection" KWD. When testing laptops over SSH, at the moment it is required to pass SNIPEIT_NO=True to the run.sh script, so that keywords dont check against snipeit when testing Novacustom devices.

JakubRed, 2025-04-08:
@SebastianCzapla You're right, I had SNIPEIT_NO in the previous run commands, and there were no errors then. I've added it again now, and the tests are passing.

SebastianCzapla, 2025-04-08:
There is PR addressing this issue, https://github.com/Dasharo/open-source-firmware-validation/pull/713
We could probably close this issue, as it has both a workaround, and ongoing fix effort.

SebastianCzapla, 2025-04-09:
Closing as https://github.com/Dasharo/open-source-firmware-validation/pull/713 has been merged.

closed 2025-04-09 by SebastianCzapla