Dasharo/open-source-firmware-validation #766: Display Detection Failure

SebastianCzapla, 2025-04-07:
### Device

vp2430

### RTE version

--

### OSFV version

develop

### Brief summary

Displays are not detected by automated tests

### How reproducible

100%

### How to reproduce

Run dasharo-compatibility/display-ports-and-lcd-support.robot on lab mounted vp2430

### Expected behavior

At least one display, the currently connected one is detected

### Actual behavior

Tests end with FAIL

### Link to screenshots or logs

[display-ports-and-lcd-support.robot_.tar.gz](https://github.com/user-attachments/files/19633709/display-ports-and-lcd-support.robot_.tar.gz)

JakubRed, 2025-04-14:
I ran these tests on v0.9.0rc6 after reinstalling Ubuntu, and all of them passed.
[PASS_display-ports-and-lcd-support_.zip](https://github.com/user-attachments/files/19734839/PASS_display-ports-and-lcd-support_.zip)

matmacieje, 2025-04-14:
Tested on Protectli VP6650, Dasharo v0.9.0, no reproduction.

[display-ports-and-lcd-support_2025_04_14_16_40_00.zip](https://github.com/user-attachments/files/19737624/display-ports-and-lcd-support_2025_04_14_16_40_00.zip)

miczyg1, 2025-04-15:
I also don't see a reason why the tests would fail, the HW works correctly:

```
ls /sys/class/drm/
card0       card0-DP-2  card0-HDMI-A-1  card0-HDMI-A-3  renderD128
card0-DP-1  card0-DP-3  card0-HDMI-A-2  card0-HDMI-A-4  version
```

Even without any display connected. Maybe the driver had not enough time po populate the sysfs entries? Given that robot can log in to the system very quickly, it is not so unlikely.

closed 2025-04-15 by matmacieje