Dasharo/open-source-firmware-validation #768: Recovery Popup is displayed but RF fails to detect it.

SebastianCzapla, 2025-04-07:
### Device

vp2430

### RTE version

--

### OSFV version

develop

### Affected component(s) or functionality

 VBO009.001 test

### Brief summary

 VBO009.001 fails, even though the tested functionality is works

### How reproducible

100%

### How to reproduce

run VBO009.001

### Expected behavior

test passes

### Actual behavior

test fails, even tho the screen is shown

### Link to screenshots or logs

[dasharo-security.zip](https://github.com/user-attachments/files/19634512/dasharo-security.zip)

matmacieje, 2025-04-14:
[verified-boot_2025_04_14_10_29_53.zip](https://github.com/user-attachments/files/19734292/verified-boot_2025_04_14_10_29_53.zip)

closed 2025-04-14 by matmacieje

BeataZdunczyk, 2025-04-24:
https://github.com/Dasharo/open-source-firmware-validation/pull/787