Dasharo/open-source-firmware-validation #772: SSH connection not configured for windows

JakubRed, 2025-04-08:
### Device

nowacustom

### RTE version

none

### OSFV version

develop

### Brief summary

Robot tests cannot connect to windows via ssh

### How reproducible

100%

### How to reproduce

run any automated windows test on novacustom v540tu 

### Expected behavior

Correct ssh connection alowing testing

### Actual behavior

Connecting to DUT on windows11 via ssh doesn't work (error:  `OSError: Socket is closed`). The keyword `Get Connections` shows that last (failed) ssh connection is: `index=1 host=192.168.4.226 alias=None port=22 timeout=30 seconds newline= prompt=root@3mdeb:/home/ubuntu# term_type=vt100 width=400 height=100 path_separator=/ encoding=UTF-8 escape_ansi=True encoding_errors=strict`.  Despite booted system being win11.

### Link to screenshots or logs

[AUD001.301 SSH_FAIL.zip](https://github.com/user-attachments/files/19650848/AUD001.301.SSH_FAIL.zip)

### Solutions you've tried

Ssh configuration form preseeds.
Manual ssh operations eg. update and restart sshagent.

wiktormowinski, 2025-04-09:
oh yeah the IPs changed and only protectlis have it noted on snipeit, it was a quick and easy system up untill this point where the IP reset and it no longer works,

for example on V1610:
```
(venv) wmowinski@fedora:~/open-source-firmware-validation$ robot -L TRACE -v rte_ip:192.168.10.218 -v device_ip:192.168.10.129 -v config:protectli-v1610 -v fw_file:protectli_v1610_v0.9.3.rom -v snipeit:no -t "AUD001.301*" dasharo-compatibility/audio-subsystem.robot 
==============================================================================
Audio-Subsystem                                                               
==============================================================================
AUD001.301 Audio subsystem detection (Windows) :: Check whether th... | FAIL |
SSH: Unable to connect - The platform may be in Windows "Recovery Mode" - Rebooted 2 times.
------------------------------------------------------------------------------
Audio-Subsystem                                                       | FAIL |
1 test, 0 passed, 1 failed
==============================================================================
Output:  /home/wmowinski/open-source-firmware-validation/output.xml
Log:     /home/wmowinski/open-source-firmware-validation/log.html
Report:  /home/wmowinski/open-source-firmware-validation/report.html
(venv) wmowinski@fedora:~/open-source-firmware-validation$ robot -L TRACE -v rte_ip:192.168.10.218 -v device_ip:192.168.10.61 -v config:protectli-v1610 -v fw_file:protectli_v1610_v0.9.3.rom -v snipeit:no -t "AUD001.301*" dasharo-compatibility/audio-subsystem.robot 
==============================================================================
Audio-Subsystem                                                               
==============================================================================
AUD001.301 Audio subsystem detection (Windows) :: Check whether th... | FAIL |
'Name                                           Status
----                                           ------
High Definition Audio Controller               OK    
HD Audio Driver for Display Audio              OK    
HD Audio Driver for Display Audio              OK    
HP ZR2440w (HD Audio Driver for Display Audio) OK    ' does not contain 'TBD_variable_not_set_03626549a59abf648ee59163b3b8acbf66c36513cb1e76d6e277bc044c926e30'
------------------------------------------------------------------------------
Audio-Subsystem                                                       | FAIL |
1 test, 0 passed, 1 failed
==============================================================================
Output:  /home/wmowinski/open-source-firmware-validation/output.xml
Log:     /home/wmowinski/open-source-firmware-validation/log.html
Report:  /home/wmowinski/open-source-firmware-validation/report.html

```
notice the `device_ip` was 129(still on snipeit) and now its 61

wiktormowinski, 2025-04-09:
hold on this issue is actually on novacustom, ill look into it further then

miczyg1, 2025-04-10:
There might be no reservations for all devices, even if the IP is on SnipeIT.

wiktormowinski, 2025-04-10:
@mkopec do we even have support for automated windows tests on novacustom laptops? there is no power control over ssh.

i know we did them on protectli or any other sonoff/rte augmented platform where you can initiate reboot in windows

wiktormowinski, 2025-04-15:
btw i was looking for the reason the keyword `Boot System Or From Connected Disk` is launching linux first and found this context: https://github.com/Dasharo/open-source-firmware-validation/pull/723/files 
this was supposed to work by launching linux first and then it should have swithced to windows automatically 
It didnt work for me though hence I did these simple changes to make it work the old fashioned way - one OS at a time.

wiktormowinski, 2025-04-15:
it works today on asset 00122, and yeah i managed to reproduce this issue @JakubRed the problem i observed was:
The test was trying to write this command while os was still booting thus i think we need to extend the `Sleep` period in that to make it work more reliably, also the lack of the actual login kwd was fixed: https://github.com/Dasharo/open-source-firmware-validation/pull/790

now it fails because of the conditions not met and not from connection issue :D
[log.zip](https://github.com/user-attachments/files/19757661/log.zip)

closed 2025-05-07 by BeataZdunczyk