Dasharo/open-source-firmware-validation #786: OS partitions to DUT drive cloning, Stage One: PoC in robot framework

matmacieje, 2025-04-11:
### The problem you're addressing (if any)

Time-consuming OS installation & configuration in Dasharo release & debug testing cycles;
configuration management issues.

### Describe the solution you'd like

DISCLAIMER: This is not going to affect or cover any OS installation tests.

I'd like to have fool-proof test suite, able to burn-in Windows, Ubuntu & Fedora on DUTs drive with at least network interfaces enabled and working SSH server. This could be done by preparing Linux system, either keeping shrinked partition images locally (but preferrably on separate partition!) or downloading them via network, dd-ing them to target partitions 0-2 (Windows, Ubuntu, Fedora) and expanding filesystems. Partition layout could be fixed with target partitions assigned to OSes, target Linux systems could share swap partition. EFI partition should be kept intact, if possible.

Shrinked OS images should contain set of drivers & FW files for typical NICs existing in our Lab. This test suite should be designed with RTE-less configurations kept in mind, so at least semi-manual execution would be possible with SSH-only DUT connectivity (i.e. notebook on the desktop).

### Where is the value to a user, and who might that user be?

Avoiding time-consuming OS installation & configuration in majority of test cases, easy & cheap way to regain OS environment stable state, whenever required in testing or debug process.

Everyone who ever needs to run some OSFV tests on either Lab platform, or on the desktop.

### Describe alternatives you've considered

Drive duplication on either dedicated hardware piece or Clonezilla workstation with all required interfaces.

miczyg1, 2025-04-11:
Certainly, the advanatge is that one can very quickly bring the setup to a usable state from the automated tests perspective.

The disadvantage, however, is the sole possibility to avoid installation of the OSes. We have to test those as well. If the possibility is given, it can be easily abused, so we have to be careful how and when we use such mechanisms.

That still would be many GBs of data to be transferred from one disk to another. if that would be an USB, then the performance wouldn't give too much time savings probably.

miczyg1, 2025-04-11:
> How to install Windows 11 on single partition?

One of these `[ TARGET 0 ][ TARGET 1 ][ TARGET 2 ]` would have to be an unallocated space (no partition or filesystem) initially. Then Windows installer will make required partitions itself.  Then one may use dd to dump the `TARGET X` part of the disk (dd can do it by specifying `skip` and `count` parameters and calculate the starting and ending block).

Regarding the checklist:

> Prepare SATA drive with required partitions for burning linux and 3 targets + EFi & swap

SWAP is not used commonly nowadays, especially with larger RAM capacities present today, so probably can be skippe,d but can be checked... Also not sure if swap can be shared across distros, it is OS RAM content TBH.

> (optional) Adjust EFI partition contents & GRUB(s)

OSes may have their own `/EFI/BOOT/BOOTX64.efi` files. These will restore the OS boot option in the  UEFI variables. There may be some conflicts between OSes/distros.

> Shrink each partition size (but not too much), prepare new images, try dd of=TARGET & filesystem expansion

This operation will be arbitrary. Someone will need to install some additional drivers or updates after burning and easily run out of space. It would probably result in frequent preparation of the base image for burning over and over again.

philipanda, 2025-04-11:
Maybe returning to maintaining the [preseeds](https://github.com/dasharo/preseeds) would be enough to solve the issue with time consuming installations?

pietrushnic, 2025-04-12:
In my previous professional life, we used https://clonezilla.org/ very heavily for that purpose. It can be netbooted from netboot.xyz, and if you point to the correct NFS, it can fetch images from some location.

DaniilKl, 2025-06-26:
Related issues and PRs:
* https://github.com/Dasharo/preseeds/pull/16
* https://github.com/Dasharo/open-source-firmware-validation/pull/274
* https://github.com/Dasharo/open-source-firmware-validation/pull/338
* https://github.com/Dasharo/preseeds/pull/18