Dasharo/open-source-firmware-validation #798: No return to Ubuntu after windows test

JakubRed, 2025-04-17:
### Device

00121

### RTE version

none

### OSFV version

develop

### Affected component(s) or functionality

novacustom laptops

### Brief summary

Windows test don't return to ubuntu afrer ending so only windows first test can pass.

### How reproducible

100%

### How to reproduce

Run two consecutive windows test after each other.

### Expected behavior

Next test boots to Ubuntu and then can be switched to windows form there.

### Actual behavior

There are no way to reboot windows, at the start or end of the test, implemented. On the novacustom platforms.
After the first windows test laptop stays in windows making next test fail.

### Link to screenshots or logs

[tpm-support_win_fai.zip](https://github.com/user-attachments/files/19791952/tpm-support_win_fai.zip)

philipanda, 2025-04-24:
Rebooting from Windows back to Linux is *sometimes* troublesome, because Windows sometimes changes the bootorder permamently, so that it's the default boot option, and changing it back from powershell is not implemented, and from I've checked briefly, not possible with stock installation (all boot entries have the same GUID).

But sometimes Windows doesn't change the bootorder, in this case [a simple reboot allows to get back to the default boot linux](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/lib/options/options-lib_dcu.robot#L176-L179). The linked lines in some way fix the issue, but don't always work and cause unnecessary boots to Ubuntu when trying to boot from Windows to Windows.

To fix the unnecessary reboots [this check](https://github.com/Dasharo/open-source-firmware-validation/blob/develop/lib/options/options-lib_dcu.robot#L188-L191) should be placed before the windows check.

The issue that Windows sometimes changes the bootorder persists and should be checked, because lately the changes were not happening like it was historically. Maybe the newest Windows update doesn't do that anymore.

macpijan, 2025-10-15:
I am not sure what we can do here. Windows bootloader is known from forcing it as the 1st boot entry by itself. Maybe the real problem is that we switch too much between Ubuntu - Windows in a single run, and we should go with full Windows run, then full Ubuntu run, etc?

philipanda, 2025-10-15:
It's just the boot order

philipanda, 2025-11-03:
Update, booting Windows does not reset the **whole** bootorder. It only resets the ESP scanning boot entries (like `Ubuntu (on SSDPR-PX700-01T-80)`, all other entries are left where they should be.

Tested by creating a custom bootentry for Ubuntu using efibootmgr, moving the two of them to the top of the bootorder and booting Windows. Only the custom one stays at the top, while the other one goes back to the end of the list.

This allows to make an easy fix by creating custom bootentries for every OS, but also shows that it could really be an issue with the firmware.

SergiiDmytruk, 2025-11-07:
I'm not sure what's the issue here.  I thought the tests typically used boot menu to select a particular OS or they don't do that for resets?

Don't know what's the logic used by Windows Boot Manager (WBM) to sort the entries, but firmware also sorts it and may interfere.  If I remember correctly, EDK was putting automatically-generated entries last the first time boot menu is opened or something like that.  Maybe there is another condition when it performs sorting.

There may be a way of preventing WBM from messing with the boot order by using `bcdedit` to remove path to the bootloader (so WBM's binary), but if tests assume a particular entry at the top that might not help.

philipanda, 2025-11-07:
> I'm not sure what's the issue here. I thought the tests typically used boot menu to select a particular OS or they don't do that for resets?

If serial console is available, then yes. If not, then we can still test purely via SSH and booting different OSes depends on setting the boot order or bootnext variable and crossing fingers that we will be able to log in via ssh to the chosen OS after a couple seconds/minutes

SergiiDmytruk, 2025-11-08:
I haven't tried it but https://askubuntu.com/a/869099 suggest to change where `{bootmgr} path` points to, deleting `{bootmgr}` may also prevent touching firmware settings (so does setting a firmware password, but that is more likely to break tests than help).  Deleting `{bootmgr}` might affect Windows updates though (was mentioned somewhere, not sure what would be the effect).