Dasharo/open-source-firmware-validation #799: OS Support - minimal test matrix

SebastianCzapla, 2025-04-17:
# Goal

Creation of a list of tests that can clearly indicate basic OS functionality is there, such as common peripherals, PC components, basic connectivity.

# List of os-agnostic tests
## Dasharo Compatibility

    CPU001 - CPU Works
    CPU002 - CPU Cache enabled
    CPU003 - Multiple CPU Support
    CPU004 - Multiple-core support
    EFI001 - Boot into UEFI OS (Ubuntu/Windows/Fedora/etc)
    NVM001 - NVMe support in OS 
    SAT001 - SATA support in OS 
    UDT001.001 - USB detection after coldboot
    UDT001.002 - USB detection after warmboot
    UDT001.003 - USB detection after system reboot
    USB001 - devices detected by OS 
    USB002 - USB keyboard in OS 
    WLE001 - Wireless card detection
    WLE002 - Wi-Fi scanning
    DSP001 - Internal display in OS (for NCM)
    DSP002 - External HDMI display in OS
    DSP003 - External DP display in OS 

## Dasharo Performance

    STB001 - Verify if no reboot occurs in the OS 

## Dasharo Security

    TPM001 - TPM Support
    TPM002 - Verify TPM version
    TPM003 - Check TPM Physical Presence Interface

## Dasharo Stability

    SNV001 - NVMe detection after cold boot
    SNV002 - NVMe detection after warm boot
    SNV003 - NVMe detection after reboot
    SNV004 - NVMe detection after suspension
    NET002 - Net controller after warmboot
    NET003 - Net controller after reboot
    NET004 - NET controller after suspend

# List of OSes to test generic functionality

    Windows 11
    Ubuntu LTS
    Ubuntu Server LTS
    FreeBSD 
    Fedora
    OPNsense
    pfSense/ pfSense CE
    OpenWRT
    Proxmox
    xcp-ng
    Debian
    ESXI
    Rocky linux

philipanda, 2025-04-24:
The `utils/basic-platform-setup.robot` was created to accomplish some of the mentioned functionality, but it hasn't been updated for a while. Maybe the suite could be extended and improved using the suggestions from this issue

SebastianCzapla, 2025-04-29:
github.com/Dasharo/open-source-firmware-validation/pull/808