Dasharo/open-source-firmware-validation #813: No description for hibernation tests

JakubRed, 2025-04-29:
### Component

Dasharo firmware

### Device

NovaCustom V54 14th Gen

### Dasharo version

-

### Dasharo Tools Suite version

-

### Test case ID

HBN001.001, HBN001.002

### Brief summary

There is no description provided for manual tests: "HBN001.001 Platform hibernation and resume (Ubuntu)" and "HBN001.002 Platform hibernation and resume (Windows)"

### How reproducible

100%

### How to reproduce

Search for HBN001.001 or HBN001.002 in Docs

### Expected behavior

proper description provided

### Actual behavior

no proper description provided

JakubRed, 2025-05-05:
Test steps could look like this:
# Manual test steps:
#    1. Power on the DUT
#    2. Login to Linux
#    3. Open terminal window and execute following command: "systemctl hibernate"
#    4. When the machine finishes hibernation process, wake it up by pressing the
#    power button

# Expected result:
#    1. The DUT should power back on
#    2. All previously opened windows should remain open

JakubRed, 2025-05-22:
The `HBN001.001` test is in the code: https://github.com/Dasharo/open-source-firmware-validation/blob/main/dasharo-compatibility/platform-hibernation-and-resume.robot
The `HBN001.002` test is not implemented anywhere.

matmacieje, 2025-06-17:
We need instruction and/or Ubuntu preseed modification to include hibernation enabling in OS.