Dasharo/open-source-firmware-validation #820: cpu-temperature measurement not possible on pcengines platform (by current osfv standard that is)

wiktormowinski, 2025-05-05:
### Device

pcengines apu2

### RTE version

-

### OSFV version

apu_rebase_0.9.1-rc2, 1884561772a2a561d91045ef80917a2bd46656f2

### Affected component(s) or functionality

`cpu-temperature` (CPT) test suite on pcengines platforms

### Brief summary

lm-sensors doesnt list temperature for cpu causing the test to fail

### How reproducible

100%

### How to reproduce

run 
```
RTE_IP=192.168.10.172 CONFIG=pcengines-apu2 DEVICE_IP=192.168.10.98 ./scripts/run.sh dasharo-performance/cpu-temperature.robot 
```

or manually on platform:
```
sensors 2>/dev/null
```
(look for `Package Id X:` or `Core X:`)

### Expected behavior

other platforms (for example protectli vp3230) have listed every single core 
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +25.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +23.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +23.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +23.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +25.0°C  (high = +105.0°C, crit = +105.0°C)
Core 4:        +21.0°C  (high = +105.0°C, crit = +105.0°C)
Core 5:        +21.0°C  (high = +105.0°C, crit = +105.0°C)
Core 6:        +21.0°C  (high = +105.0°C, crit = +105.0°C)
Core 7:        +21.0°C  (high = +105.0°C, crit = +105.0°C)

nvme-pci-0500
Adapter: PCI adapter
Composite:    +25.9°C  (low  =  -5.2°C, high = +82.8°C)
                       (crit = +84.8°C)
Sensor 1:     +30.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +25.9°C  (low  = -273.1°C, high = +65261.8°C)

spd5118-i2c-2-50
Adapter: SMBus I801 adapter at efa0
temp1:        +24.0°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)
```

### Actual behavior

the sensors output shows some pci devices and there are no `Package Id` or `Cores` in sight:

```
ubuntu@3mdeb:~$ sudo sensors 2>/dev/null
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +42.9°C  (high = +70.0°C)
                       (crit = +105.0°C, hyst = +104.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +42.8°C  

ath10k_hwmon-pci-0100
Adapter: PCI adapter
temp1:        +24.0°C  

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:      976.00 uW (interval =   0.01 s, crit =   6.00 W)

```

is one of them an average CPU temperature by any chance?????

### Link to screenshots or logs

[cpu-temperature_log.zip](https://github.com/user-attachments/files/20040366/cpu-temperature_log.zip)

miczyg1, 2025-05-06:
@wiktormowinski AMD platforms are different than Intel. You can't expect everything to be the same. 

On AMD the driver is `k10temp` and it shows the temperature of whole CPU package. Why can it not be used to measure the temperature? Is it impossible to take `temp1` as the CPU temperature?

For comaprison, on Intel the package temperature is equal to the highest temperature on any core.

wiktormowinski, 2025-05-06:
thanks for confirming @miczyg1, now that this is clear we can work on a fix for that.