Dasharo/open-source-firmware-validation #823: Apu2 | VBO suite makes the platform not responsive (probably bricked) | v0.9.1.-rc2

wiktormowinski, 2025-05-06:
### Device

pcengines apu2

### RTE version

-

### OSFV version

apu_rebase_0.9.1-rc2, 1884561772a2a561d91045ef80917a2bd46656f2

### Brief summary

Launching `verified-boot` suite causes a brick at some point in the middle of testing 

### How reproducible

Happened to me consistently. However, cannot grant it as a full 100%

### How to reproduce

```
RTE_IP=192.168.10.172 FW_FILE=dump_apu_two.rom CONFIG=pcengines-apu2 DEVICE_IP=192.168.10.98 ./scripts/run.sh dasharo-security/verified-boot.robot 
```
Note: this fw is just a read dump of 0.9.1-rc2 with ubuntu as boot option.

### Expected behavior

individual run of these test cases passes so i expect the whole suite to do so as well 

### Actual behavior

it fails every single test case past VBO008.001, and the platform doesnt print anything on serial. 
Flashing the fw again helps, thus launching these test cases individually works just fine. 

### Link to screenshots or logs

[vbo-fail-mid-testinh.zip](https://github.com/user-attachments/files/20067140/vbo-fail-mid-testinh.zip)

macpijan, 2025-05-09:
@wiktormowinski is it related: https://github.com/Dasharo/open-source-firmware-validation/issues/250? Do you search through issues prior submitting a new one?

wiktormowinski, 2025-05-09:
this issue marks something different, one of these seemingly random fail in the middle of the suite and leaving the platform in a state that needs a reflash. At the other hand we got this https://github.com/Dasharo/open-source-firmware-validation/issues/250 which is focused solely on VBO009.001 failing. I deem it as different enough.