Dasharo/open-source-firmware-validation #831: No "Secure boot" in dmesg, despite Secure boot up and running

matmacieje, 2025-05-12:
### Device

NovaCustom V560TND

### RTE version

not used

### OSFV version

develop

### Affected component(s) or functionality

OSFV, SBO002.201 UEFI Secure Boot (Ubuntu)

### Brief summary

No "Secure boot" in dmesg, despite Secure boot up and running

### How reproducible

10/10

### How to reproduce

1. Enable Secure boot
2. Boot Linux (Ubuntu 24.10, kernel 6.11.0-25-generic)
3. sudo dmesg | grep "Secure boot"
4. Verify if secure boot actually works (i.e using hello-dasharo program)

### Expected behavior

"Secure boot" string existing in dmesg (or new detection method).

### Actual behavior

No "Secure boot" in dmesg, despite Secure boot up and running

### Link to screenshots or logs

No log, manual testing.

miczyg1, 2025-05-13:
@matmacieje  Upper/lower case matters. Surprise, surprise...

```bash
sudo dmesg |grep secureboot
[sudo] password for miczyg: 
[    0.000000] secureboot: Secure boot disabled
[    0.008280] secureboot: Secure boot disabled
```

matmacieje, 2025-06-02:
Final conclusion: to avoid situations like this, test should be redesigned to use journalctl instead of dmesg.

closed 2025-06-26 by filipleple