Dasharo/open-source-firmware-validation #837: No automation for AUD001.202 and AUD004.202 (Fedora), no description in docs

matmacieje, 2025-05-14:
### Device

NovaCustom V560TND

### RTE version

None

### OSFV version

develop

### Affected component(s) or functionality

dasharo-compatibility/audio-subsystem.robot

### Brief summary

No automation for AUD001.202 and AUD004.202 (Fedora)

### How reproducible

10/10

### How to reproduce

Try to execute tests AUD001.202 and AUD004.202 on Fedora

### Expected behavior

AUD001.202 and AUD004.202 test cases implemented in dasharo-compatibility/audio-subsystem.robot

### Actual behavior

AUD001.202 and AUD004.202 test cases not implemented dasharo-compatibility/audio-subsystem.robot

### Link to screenshots or logs

No.

### Solutions you've tried

Executed tests manually, using Ubunt (.201) robot code as a reference.

philipanda, 2025-05-21:
The tests were added in https://github.com/Dasharo/open-source-firmware-validation/commit/d87ec2f5fa2b1e14f8664ff92fb2e683dbed7ab6
and then removed in https://github.com/Dasharo/open-source-firmware-validation/commit/bd5158f19e24.

philipanda, 2025-05-26:
The Fedora tests are back again and work like the author of the suite overhaul intended.
The tests regarding external headset fail in the end due to a headset not being connected to the device.
[audio-subsystem_log.zip](https://github.com/user-attachments/files/20435840/audio-subsystem_log.zip)