Dasharo/open-source-firmware-validation #864: Update pre-commit linters, especially robocop

m-iwanicki, 2025-06-05:
### The problem you're addressing (if any)

I noticed couple bugs with how robocop capitalizes variables.

### Describe the solution you'd like

Updating to 5.8.0 fixes problems I encountered, it just requires adding

```toml
# To allow setting e.g. DUT_CONNECTION_METHOD
"0929:enabled:False"
```

to `robocop.toml`.
There is version v6.3.0 as of writing this but it requires migrating config as there was major change with 6.0.0 release. From this version onwards robocop and robotidy were combined into one tool.

Updating shouldn't be big problem (except maybe more pre-commit errors to fix/ignore) but there is tool synchronization problem with pre-commit and `requirements.txt`. In `requirements.txt` we currently use robocop 5.0.4. If it isn't needed in virtual environment then it would be easiest to just remove it and run pre-commit robot hook if needed. Trying to update robocop in venv to version `6.*.*` might be problematic as there are dependency conflicts e.g. `robotframework-sherlock` requires [pathspec==0.9.*](https://github.com/MarketSquare/robotframework-sherlock/blob/eae11f65eb10dd8c7ea9825292fe59eff0baa962/setup.py#L48) while robocop requires [pathspect>=0.12](https://github.com/MarketSquare/robotframework-robocop/blob/c21cfa578abf0464eaa1e42fe21e1d06f05051ae/pyproject.toml#L20)

### Where is the value to a user, and who might that user be?

User: test developers and reviewers
Value: Better linting, error detection

macpijan, 2025-06-20:
I think it was supposed to be fixed by: https://github.com/Dasharo/open-source-firmware-validation/pull/827

JakubRed, 2025-06-26:
Current version of Robocop in OSFV has been set to v6.5.0: https://github.com/Dasharo/open-source-firmware-validation/pull/827

closed 2025-06-26 by BeataZdunczyk