Dasharo/open-source-firmware-validation #886: Feature array for test cases

krystian-hebel, 2025-06-13:


philipanda, 2025-06-18:
> It would be an additional array per case in the test_cases.json file, as well as an [RF tag](https://robotframework.org/robotframework/latest/RobotFrameworkUserGuide.html#tagging-test-cases) in the test definitions.

I wonder if seperating the features from plain tags in the database, but keeping them together as tags in tests won't cause issues. That would only work if we decided to never keep other tags in the database, and still it would make verifying if all the features listed in a test case are propagated to the database a pain.

Maybe we should just save all of the tags of the test cases to the database?
That would also include tags related to automation, as we planned to add some kind of `manual` or `semi-auto` tags.

Alternatively we could come up with some tagging convention, for example, features could look like `feature:tpm_support`, `feature:audio`. That way we could differentiate the feature tag from everything else.

philipanda, 2025-06-18:
@krystian-hebel I've been discussing this issue with @macpijan and we are not sure if adding tags to every test case in OSFV will be the best option.
Aren't the tests for given features more or less categorized into suites?
Couldn't we say that a test suite is testing one feature? It's the case most of the time. If some suites are too big, then we could split them.
There are on some occasions features, that should trigger a couple of suites, but I can't think of any feature that would trigger only single test cases from a suite. Wouldn't it be better to instead categorize the tests by suites? Not one test case by one test case?

philipanda, 2025-06-23:
@krystian-hebel What about using variables for specifying features support?
We've been using them for that already in conjunction with the `Skip If` keyword, but the use was not documented and the variables were often not maintained properly resulting in a huge amount of config variables.

Skip config variables could serve the exact same use as tags, and could even help us with the issue of platform-configs almost always needing changes during testing. Instead of specifying a list of tags to be performed, we could specify a list of feature skip variables to be set to ${TRUE}. The list could be appended to the platform config file, or just imported as a resource, to override all of the skip variables with given values.

krystian-hebel, 2025-06-23:
I wouldn't say that they are used for the same role. Variables specify what is supported by the platform (both because of platform itself, as well as by its configuration in lab, like PiKVM vs just serial), but it doesn't necessarily mean that everything that is supported will be part of every release for that platform.

> the issue of platform-configs almost always needing changes during testing

I'm not aware of this, why it has to be changed so often?

> The list could be appended to the platform config file, or just imported as a resource, to override all of the skip variables with given values.

Please don't, this would mean that each test is executed from `dirty` git tree.

Both tags as well as variables can be passed to RF in command line parameters. This could get very long if most of the config options are switched from their defaults, but it is also something that can be easily automated.

philipanda, 2025-06-23:
> I wouldn't say that they are used for the same role

I don't mean that they are equivalent, but I believe there are potentially many variables that actually show a release scope, not hardware capabilities.
For example are `${CUSTOM_BOOT_MENU_KEY_SUPPORT}`, `${CAPSULE_UPDATE_SUPPORT}`, or `${FIRMWARE_NUMBER_VERIFICATION}` something hardware specific? Or only a decision, that we want to add this feature in a given release?

> I'm not aware of this, why it has to be changed so often?
A few examples:
- `${DEVICE_USB_KEYBOARD}` - is changed on every release because the platforms that support USB keyboards are not always set up with a dedicated keyboard. The keyboard model should be defined at the hardware matrix at docs.dasharo, but finding it and then finding the required keyboard is often much more troublesome than just connecting any keyboard and saying the test passes without uploading the logs, or just modifying the model in configs.
- `${WEBCAM_UBUNTU}`, `${BLUETOOTH_CARD_UBUNTU}` etc. - sometimes the exact names differ depending on OS or their versions and are often updated, similar to DEVICE_USB_KEYBOARD
- `${DMIDECODE_RELEASE_DATE}` - must be changed on every release, there is nothing to do with that other than reading the value from the firmware image used for testing - a whole other topic of automatic handling of the tested firmware image
- `${TESTS_IN_WINDOWS_SUPPORT}`, `${TESTS_IN_UBUNTU_SUPPORT}`, `${TESTED_LINUX_DISTROS}` - the two variables are used to tell robot to perform the tests in only Windows or Ubuntu.

> Please don't, this would mean that each test is executed from dirty git tree.

I don't think any file managed by git has to be modified for that. It should be possible to read a filename containing the release scope variables from an environment variable and load them conditionally if provided. The file should be generated by osfv-dashboard or some other tool based on the test scope and distributed to the testers. The contents, or a hash of this file could easily be logged and read by osfv-dashboard to verify if the test scope file was not modified compared to what the test scope at osfv-dashboard defines.

> Both tags as well as variables can be passed to RF in command line parameters

Only that tags can only be passed as command line parameters, while the variables can be defined in a file, path to which could be given as an optional parameter or environment variable.

> This could get very long if most of the config options are switched from their defaults, but it is also something that can be easily automated.

I don't think it being long would be an issue if we could automate it, like you said.
The platform configs are long and hard to maintain as of today, and they are not generated automatically.