Dasharo/open-source-firmware-validation #934: phoronix cachebench result outside of acceptable range

matmacieje, 2025-07-09:
### Component

other

### Device

NovaCustom V56 14th Gen

### Dasharo version

1.0.0-rc3

### Dasharo Tools Suite version

none

### Test case ID

UPP001.302

### Brief summary

Phoronix Test Suite: cachebench (Test: Read / Modify / Write) score is: 139443.160207 MB/s
which is out of range defined in platform-config: 130156.07999999999 - 86770.72

### How reproducible

3/3  (several re-runs, always out of range)

### How to reproduce

Under Windows 11on NovaCustom V560TND, install Phoronix Test Suite, install and run cachebench test, check average results of last (Read / Modify / Write)) sub-test

### Expected behavior

Benchamrk result in projected range.

### Actual behavior

Benchamrk result out of range.

### Screenshots

[WINDOWS_manual_cpu_performance_2025_05_26_14_38_12.zip](https://github.com/user-attachments/files/20441693/WINDOWS_manual_cpu_performance_2025_05_26_14_38_12.zip)

### Solutions you've tried

Couple of re-runs.

filipleple, 2025-08-25:
NUC BOX: `cachebench` test failed because the result `106987.014884` is `out of acceptable values: 94153.92 - 62769.280000000006`, i.e. too good. The result roughly corresponds with 14th gen CPU results. Same for `smallpt`.

philipanda, 2025-08-29:
Looks like the creator of this test suite just got very low scores on the benchmarks when deciding on the reference values for some reason.

For example on [openbenchmarking.org ](https://openbenchmarking.org/result/2505075-NE-20250507035) we can see that recently someone got a score nearly twice as good on smallpt than the one defined for V540TU (~15s vs ~27s)

I wonder which approach to solving this issue would be the best?
- Update the scores a bit to better reflect the expected performance
- Increase the tolerances - that can make this test much less useful and experience false positives
- Only respect the lower tolerance - so that a score higher than expected is always accepted
  - some tests have a reverse scale, like smallpt - the lower the score the better

philipanda, 2025-08-29:
I think the best approach would be to:
- only FAIL if the performance is unexpectedly LOW
  - if it's higher than expected a WARN could be issued, but that shouldn't cause FAILs
- the reference values must be updates
  - openbenchmarking.org publishes average scores from the reports published by the community for benchmarks on a given hardware. This site might be a good reference point eg. https://openbenchmarking.org/test/pts/smallpt

philipanda, 2025-08-29:
The issue is fixed by this PR
https://github.com/Dasharo/open-source-firmware-validation/pull/1049

closed 2025-09-04 by BeataZdunczyk