Dasharo/open-source-firmware-validation #967: NUC Box random characters printed in setup menu

philipanda, 2025-07-28:
### Component

Dasharo firmware

### Device

NovaCustom NUC BOX 14th Gen

### Dasharo version

v0.9.0-rc2

### Dasharo Tools Suite version

-

### Test case ID

Any automatic test case in setup menu

### Brief summary

A few random characters are sometimes printed in random places. This completely wrecks automatic tests via serial console. It's not very noticeable most of the times when traversing it manually.

### How reproducible

~5% per "frame" 100% if you move around the menu for a whie and observe carefully

### How to reproduce

Run some tests in setup menu or connect to the plaform via serial and slowly move around, as refreshing the screen will most often remove the artifact

### Expected behavior

No random characters

### Actual behavior

Check the logs at:
Run Keywords > Restore Secure Boot Defaults > Enter Secure Boot Menu And Return Construction > Enter Secure Boot Menu > Enter Setup Menu And Return Construction  

### Screenshots

No newline in help text
[image]

`1D` printed after the setup menu
[image]

Random `@` in the middle
[image]

A total mess
[image]

### Additional context

The platform uses an USB-FTDI converter. Maybe it's the converters fault?
It was the #00245 

### Solutions you've tried

Rebooting and coldbooting didn't help.

filipleple, 2025-07-28:
It's not only on the NUC BOX, all FTDI-reliant platforms are prone to this. This seems to be due to the instability of FTDI/other UART-USB converter combinations, and is more of a problem with our testing setup, not something we can resolve in firmware

pietrushnic, 2025-07-28:
This is an ancient problem (at least 15+ years to me), and I doubt there is a good solution to it. I saw some filtering methods, but there would always be something. This would be expensive to solve and require an intense dive into the correct forum.

IIRC we experimented with other methods of encoding and other terminal emulators, which improved the situation but never fully eliminated the issue.

macpijan, 2025-07-28:
What exactly is this ancient unsolvable problem? What are the conditions for it to happen?
Because the whole Dasharo validation depeds on it.

pietrushnic, 2025-07-28:
> What exactly is this ancient unsolvable problem? What are the conditions for it to happen? Because the whole Dasharo validation depeds on serial connection.

Expecting fully reliable (no additional random characters or noise) in serial output at system bootstrap/UEFI BIOS level.

Conditions are as follows: we have a platform either with a serial (COM) output on the motherboard or a serial port exposed through a ping on the platform, then we try to leverage that connection through converter to catch output of boot log or UEFI BIOS setup menu over that serial, on other side we analyze that output to make decisions. I also had experience with the Option ROM environment.

PC Engines was made for industrial usage, and even there, we see some random stuff from time to time.

In my experience, I spent countless hours fixing and chasing issues like:

> A few random characters are sometimes printed in random places.

From my experience, it is challenging, impossible, or at least economically infeasible to solve. If we have data/stats, we can improve the reliability of serial/UART, but it would be tough to address that to eliminate random characters. There have to be some critical infrastructure/systems designs and/or standards that address that, but I didn't get to that level of analysis.

filipleple, 2025-07-29:
Might be worth noting that after transition to Ubuntu, the connection seems perfect. Considering that the regular serial port console redirection is also flawless, it most likely has to do with the internals of the FTDI driver we've integrated. 

I think that if there is a similar driver in EDK2 for the serial port, it would be useful to look at it for any extra error handling hacks and adapt them in the FTDI driver.

macpijan, 2025-07-29:
> I'm not sure what highly reproducible means

We basically cannot run any test on this platform right now.

We can run test just fine on any other platform not using this edk2 FTDI driver.