Dasharo/open-source-firmware-validation #1029: IP assigned to MSI sometimes changes depending on flashed firmware resulting in failure to boot into DTS

m-iwanicki, 2025-08-25:
### Device

MSI

### RTE version

N/A

### OSFV version

develop (ae7110d8c82f)

### Brief summary

IP assigned to DTS might change depending on which fw is flashed which results in `Boot Into DTS` keyword failing.

### How to reproduce

1. Flash different fw (release/rc/propertiary)
2. Boot into DTS
3. Check IP assigned by dhcp

### Expected behavior

Same IP every time

### Actual behavior

Sometimes it's different

### Link to screenshots or logs

-

### Solutions you've tried

Repeat test with new IP.

miczyg1, 2025-08-26:
Maybe IP is not reserved in DHCP?

m-iwanicki, 2025-08-26:
Maybe MAC changes depending on FW?

macpijan, 2025-08-26:
Yes maybe MAC has changed, maybe connected somehow via other means (if this unit has WiFi - I am not really sure). We need more details in the report to make sure.